HKUST-Aerial-Robotics/VINS-Mobile #3: 程序一直提示"wait for imu, only should happen at the beginning"

victoryli88, 2017-05-08:
iPhone 7

运行程序时，一直提示"wait for imu, only should happen at the beginning"，是需要配置么？

 if (!(imu_msg_buf.back()->header > img_msg_buf.front()->header))
        {
            NSLog(@"wait for imu, only should happen at the beginning");
            return measurements;
        }

PeiliangLi, 2017-05-08:
hi,
Please make sure using the provides opencv2.framework instead of the official version.
If you cannot access dropbox, download it from https://pan.baidu.com/s/1b9O2Ai

PeiliangLi, 2017-05-08:
use the opencv from https://pan.baidu.com/s/1o7F5FnW

PeiliangLi, 2017-05-08:
hi,
Please make sure using the provides opencv2.framework instead of the official version.
If you cannot access dropbox, download it from https://pan.baidu.com/s/1o7F5FnW

closed 2017-05-08 by PeiliangLi

match08, 2020-02-22:
> hi,
> Please make sure using the provides opencv2.framework instead of the official version.
> If you cannot access dropbox, download it from https://pan.baidu.com/s/1o7F5FnW

Undefined symbols for architecture arm64:
  "cv::Mat::updateContinuityFlag()", referenced from:
      cv::Mat::Mat(int, int, int, void*, unsigned long) in VINS.o