HKUST-Aerial-Robotics/VINS-Mobile #6: no such file 'VINS-Mobile/VINS-ios-framework/VINSFramework.mm'

TonyMou, 2017-05-08:
build failed, and show the error like 

clang: error: no such file or directory: '/Volumes/Samsung SSD T3/DobbyMac/Docs/resources/VINS-Mobile/VINS-ios-framework/VINSFramework.mm'
clang: error: no input files

PeiliangLi, 2017-05-08:
Hi,
please make sure using the provided opencv2.framework.

TonyMou, 2017-05-08:
Thanks for the reply and yes I am using the opencv u provide still not work

closed 2017-05-09 by TonyMou

hyhrob, 2017-05-26:
I'm facing the same issue. Have you resolved it? @TonyMou @PeiliangLi

TonyMou, 2017-05-26:
You need to uninstall your OSX's opencv. @hyhrob

hyhrob, 2017-05-26:
Yes, I have uninstalled it and the xcode can correctly find the header files  in opencv.framework in the thirdparty folder. But nothing changed.

hyhrob, 2017-05-31:
sorry, my fault. I have used a wrong package about opencv2.framework.
I have changed some code and it now can running on iPhone6 plus. Maybe I used the initialization parameters of iPhone6s plus, it always lost its location.

hyhrob, 2017-05-31:
Good job! @TonyMou

tigerking, 2017-06-02:
Another solution is to rename /usr/local/include/opencv2 to another folder name,eg Xopencv2. 
So you need not uninstall opencv on your MAC. when you need opencv, rename the folder name from back to opencv2.
$sudo mv /usr/local/include/opencv2 /usr/local/include/Xopencv2