HKUST-Aerial-Robotics/VINS-Mobile #7: newbee in ios

zdczdcc, 2017-05-08:
I'm a newbee in ios.I used xcode to open your project cloned from the hub.It shows some error like in the below image.

PeiliangLi, 2017-05-08:
Hi,
It does not mean an error, you can ignore them and just build VINS-ios.

zdczdcc, 2017-05-08:
I just built it,and it reported some errors.
[image]

zdczdcc, 2017-05-08:
Like that it's an error from opencv.But the `opencv.framework` i used was downloaded from dropbox instead of the official version.

PeiliangLi, 2017-05-08:
have you installed opencv in your Mac?

TonyMou, 2017-05-09:
uninstall your OS's opencv and use the opencv.framework the author provided.

zdczdcc, 2017-05-09:
@TonyMou @PeiliangLi you are all right.I uninstalled my mac's opencv and solved this problem.

closed 2017-05-09 by zdczdcc