HKUST-Aerial-Robotics/VINS-Mobile #8: ceres problem？

danpeng2, 2017-05-09:
Hi. .It shows some error like in the below image.

[image]
[image]

PeiliangLi, 2017-05-09:
Hi，
Have you chosen your device instead of simulator as the target when you built？

danpeng2, 2017-05-09:
yes， I chose simulator as the target when I built

danpeng2, 2017-05-09:
@PeiliangLi yes， I chose simulator as the target when I built

PeiliangLi, 2017-05-09:
That's the problem. You should connect your device and choose your iPhone as the target.

danpeng2, 2017-05-09:
@PeiliangLi thank you！  Compatible Devices have  iPhone6s but not  iPhone6？

PeiliangLi, 2017-05-09:
yes. We have not tested it on iPhone6

closed 2017-05-10 by PeiliangLi