HKUST-Aerial-Robotics/VINS-Mobile #17: Why limit device

gxl2024595, 2017-05-12:
Hi
   I find that the lowest device become iphone6s. The earlier versions can run on iphone6 and ipad Air 2,and i want to konw why add the limit.

PeiliangLi, 2017-05-12:
Hi,
It's for the performance guarantee. If you want to test the performance on iPhone6 or iPad Air, you can just relax the device limit on your repo.

gxl2024595, 2017-05-12:
I run the Demo on iPad Air2 , it always crash here.please tell me why it did happen usually . And could you tell me  how to reslove the problem?

[image]

PeiliangLi, 2017-05-12:
We didn't test it on iPad Air, so you need to tell me your modification in code.

gxl2024595, 2017-05-12:
[image]

i  relax the device limit, just modifi here.And the same error would present in earlier versions.

qintony, 2017-05-12:
numerical stable issue. 
For example, long time stay still, IMU propagation, the covariance matrix become bigger and bigger.  So the information matrix bigger smaller and smaller. It will cause some numerically unstable issue.

We assert here to avoid this.

gxl2024595, 2017-05-15:
Then how can I avoid this problem?

zjupx, 2017-05-18:
I also met this assert in my iPhone6P. when I just use iphone 6sP 's params.  It seems all working fine at begining, but go to this assert later.

gxl2024595, 2017-05-20:
@zjupx Hi,have you solved this problem?

closed 2017-06-07 by PeiliangLi