HKUST-Aerial-Robotics/VINS-Mobile #18: Confused about code for initial alignment

vonzy, 2017-05-13:
Hi, I am a college student from Information Engineering Univesity.
I have read your paper _Robust Initialization of Monocular Visual-Inertial Estimation on Aerial Robots_, but feel a little confused about your code for **solveGyroscopeBias**, I don't know why the matrix A and b are consturcted like that, Can you show me the matrix fomula for the following code? thanks!!! 
`        tmp_A = frame_j->second.pre_integration->jacobian.template block<3, 3>(O_R, O_BG);
        tmp_b = 2 * (frame_j->second.pre_integration->delta_q.inverse() * q_ij).vec();
        A += tmp_A.transpose() * tmp_A;
        b += tmp_A.transpose() * tmp_b;`

qintony, 2017-05-13:
That the Jacobian Matrix propagated through each integration step.

see the integration_base_h.

vonzy, 2017-05-13:
@qintony I mean I don't know how to construct the Matrix A and b for least square estimation ,for example why the code add ttmp_A.transpose() * tmp_A to A in the loop.

qintony, 2017-05-13:
To solve Ax = b. If A is a rectangle, you can not do the inverse.
A'Ax = A'b  If A'A is full rank, you can do the inverse operation and solve it.
sum (tmp_A' * tmp_A) = A'A. prove it by yourself.

vonzy, 2017-05-13:
yes! that's right! thank you very much!!!

vonzy, 2017-05-13:
@qintony but I am still not aware of why must sum tmp_A to get A, I mean it seems that tmp_Ax=tmp_b,is it right to construct matrix equation like 
[image]

qintony, 2017-05-23:
How to solve tmp_Ax = tmp_b when tmp_A is the rectangle?
tmp_A' * tmpA...      is equal to sum every one.

closed 2017-06-14 by PeiliangLi