HKUST-Aerial-Robotics/VINS-Mobile #21: iPhone 6 failing ...

pawlusmall, 2017-05-15:
Hi,

I was trying to test the project on an iPhone6 (not S), but it always fail (FAIL_IMU, FAIL_SFM...).

I changed focus lengths and PX, PY with my own but still fails.

Do I need to change any other parameter? 

Thanks and congratz for you awesome work!

PeiliangLi, 2017-05-15:
Hi, thx for attention.
The compatible devices do not include iPhone6.

pawlusmall, 2017-05-15:
Hi, I know iPhone6 is not listed as compatible device, but I think that is because the algorithm cannot guarantee enough performance (FPS) as in newer devices.

My question is, what parameters should I configure in order to make it initialize and run, even though I know it won't run as fast as iPhone6s and above?

PeiliangLi, 2017-05-15:
Check if your xcode is in the debug mode.

pawlusmall, 2017-05-15:
It is in release mode.

PeiliangLi, 2017-05-15:
We don't have iPhone6 to test. But I guess it's may because we limit the sfm solver time, so it cannot converge on iPhone6.
Find the code at https://github.com/HKUST-Aerial-Robotics/VINS-Mobile/blob/master/VINS_ios/inital_sfm.cpp  line283

pawlusmall, 2017-05-15:
Thank you, I will take a look at it and post if I find an issue with that.

pawlusmall, 2017-05-16:
Hi,

Tried to increase the initial SFM solve time but still failing. I've also reduced the number of features to 800.

Here it is my output log:

`2017-05-16 08:26:25.489161+0200 VINS_ios[596:132550] vins delay 0.322022
imu error 16482.437217 16482.431720 16482.447227
imu error 16482.447227 16482.441621 16482.457206
imu error 16482.477164 16482.471353 16482.487174
imu error 16482.487174 16482.481287 16482.497153
TIMER_time_goodfeature: 42.78ms
TIMER_time_feature: 117.91ms
adding feature points 60
size of img = 60
number of feature: 99
solve initial------------------------------------------
PS -52.866614 15.063440 -18.096583
aver_g  -8.15586 -0.335967   5.53729
IMU variation 0.791895!
init solve 5pts between first frame and last frame failed
marginalize back
TIMER_process_image: 3.37ms`

As you can see it fails in the initialization. Is that IMU error normal?

PeiliangLi, 2017-05-16:
Can you show me more results about imu log?
Btw, you should change MAX_CNT in feature_tracker.hpp to reduce the feature number other than NUM_OF_F.

pawlusmall, 2017-05-16:
Sure, here it is a full output log:

https://pastebin.com/kZ51rj91

Changed MAX_CNT to 50, working faster now but still failing.

PeiliangLi, 2017-05-16:
IMU always error. That's the problem, I suggest you check the Apple API to see if there are some differences of IMU interface on iPhone6. Also, upgrade your device to at least 10.2.1 if not.

pawlusmall, 2017-05-16:
Hi, Finally I made it work using iPhone 7S params instead of 6S.

My impressions:

- It works well under slow, medium movement.
- It drifts a little when moving faster.
- It is able to relocalize from drift but after that it sometimes jumps its position, like if it made a wrong loop closing and its still detecting a drifted keyframe.
- Relocalizing is usually very slow (5 secs or more).
- Under very hard movements sometimes it is not able to relocalize

PeiliangLi, 2017-05-16:
Thanks for your detailed test.
The wrong loop bug has been fixed, please pull the newest commit. Relocalization(global pose graph) should not be that slow unless you run it by very long time. 
We cannot detect loop under hard movements because of the serious motion blur.

closed 2017-05-18 by PeiliangLi