HKUST-Aerial-Robotics/VINS-Mobile #24: Coordinate systems

K0stIa, 2017-05-17:
Hi,

what is the slam coordinate system? 

I found in code that IMU has this one:
 Z^
 |   /Y
 |  /
 | /
 |/--------->X

Is the SLAM system same?

btw, machine hall dataset has this one, right?
      /Z
    /
   /
 |/--------->X
 |
 |
 |
Y

Thanks.

PeiliangLi, 2017-05-17:
We use the first one in VINS-Mobile.
Machine Hall Datasets were tested on ros version which runs on ubuntu and will be released soon.

K0stIa, 2017-05-17:
Both, imu & visual have same coordinate system?

PeiliangLi, 2017-05-17:
Of course not, it depends on the rotation and translation between IMU and camera. You can find the params in code.
We treat the IMU frame as the body frame

K0stIa, 2017-05-17:
yeah, could you point out where you apply this transformation?

K0stIa, 2017-05-17:
am I right that these parameters define needed transformation?
#define RIC_y ((double)0.0)
#define RIC_p ((double)0.0)
#define RIC_r ((double)180.0)
#define TIC_X ((double)0.0)
#define TIC_Y ((double)0.043)
#define TIC_Z ((double)0.0)

Thanks in advance.

PeiliangLi, 2017-05-17:
yes, right.

K0stIa, 2017-05-17:
I still have problems with running your stuff on machine hall datasets.
Would it be possible to send me parameters with which I can fly on these datasets?
-K.

K0stIa, 2017-05-17:
Currently, I am trying to use these extrinsics from the Machine Hall 01 easy dataset.

# General sensor definitions.
sensor_type: camera
comment: VI-Sensor cam0 (MT9M034)

# Sensor extrinsics wrt. the body-frame.
T_BS:
  cols: 4
  rows: 4
  data: [0.0148655429818, -0.999880929698, 0.00414029679422, -0.0216401454975,
         0.999557249008, 0.0149672133247, 0.025715529948, -0.064676986768,
        -0.0257744366974, 0.00375618835797, 0.999660727178, 0.00981073058949,
         0.0, 0.0, 0.0, 1.0]

PeiliangLi, 2017-05-17:
VINS-Mobile is on iOS platform with living video and IMU data. How did you modify it to run with Euroc dataset. It's not an easy task unless you fully understand the whole pipeline.

K0stIa, 2017-05-17:
I do not use loop closure for now. I am just trying to feed imu+timestamp and images(cam0)+timestamps into your code. I wrote separate driver for that which feeds data to VINS class from your code. (I took intrinsics and extrinsics from MH01_easy dataset).

PeiliangLi, 2017-05-22:
@K0stIa I highly recommend you to wait for our Ros-version, it will open source in 1～2 days and can reproduce the results easily without any modification.

closed 2017-05-22 by PeiliangLi

K0stIa, 2017-05-22:
thanks a lot @PeiliangLi, I am looking forward to it.

PeiliangLi, 2017-05-22:
https://github.com/HKUST-Aerial-Robotics/VINS-Mono 
**Open source now.**

K0stIa, 2017-05-22:
Thanks a lot @PeiliangLi, I will play with it. It looks really cool.