HKUST-Aerial-Robotics/VINS-Mobile #25: can't you share your opencv-framework source?

zjupx, 2017-05-17:
From Readme, you have a modified opencv framework.  I find in your code , you use this opencv framework to get Video Frame's timestamp.  Can you share this modify code?  
For my test,  the time stamp is almost faster 48ms than in  "processImage" function using systemUptime.
If I want to get the real video frame's timestamp in android. What I can do for that?

====
您好，非常感谢您能开源这么棒的资源。 我想请教一下您关于视频帧的时间戳获取的方法。
看您的代码，您修改了opencv的代码。从而获取了视频帧的真实时间戳。我实际测试，发现这个时间戳比在processImage方法中的systemUptime 提前了有48ms左右。能分享一下您获取视频帧时间戳这一块的代码吗？
另外如果我想在android中完成同样的功能。是否也能够借用这里的思路？
谢谢，期待您的回复。

PeiliangLi, 2017-05-17:
Hi,
I open the modified part in https://github.com/PeiliangLi/opencv_modify, it's because Apple does not provide the image timestamp API.
If Android can provide the API, then you don't need to hack it.

zjupx, 2017-05-17:
really appreciate for your reply.  
Android do not provide this Video frame timestamp's API.  So maybe I need some hack like yours.
I want ask Another question:
I found some IMU noise's params in global_param.hpp, like ACC_N\ ACC_W\GYR_N\GYR_W.  how do you know these params, just try, or find these from IMU's datasheet, or some other way?
If I want to port to android,  Can I use these direct?

====
非常感激您的快速回复。
Android中没有提供类似视频帧的timestamp的功能。所以可能我需要想办法自己实现了。
另外还想向您请教一个问题：
我看到在global_param.hpp中，您定义了四个IMU的噪声数据。 ACC_N\ ACC_W\GYR_N\GYR_W 。 如果没有猜错的话，ACC_N代表加速度计的白噪声，  ACC_W代表加速度计的bias的随机游走噪声， 是这样的吗？
这些参数是如何确定的呢？ 是您尝试出来的，还是从数据手册中获取到的呢？ 如果我想用在其他的手机上，这些参数是否还能通用？
再次感谢您的回复，非常有帮助。谢谢！

PeiliangLi, 2017-05-17:
They are trade-off parameters that you can tune them according to the quality of your sensors.
Basically, as for mobile platform, you can just use them because the IMU sensors have almost the same quality.

closed 2017-05-19 by zjupx