HKUST-Aerial-Robotics/VINS-Mobile #26: Unable to link to ceres-solver library

shubh-agrawal, 2017-05-17:
Hi, I am having some problem while compiling your project in Xcode 8. The error messages convey to me that the project is unable to link to ceres-solver and also has "Undefined symbols for architecture x86_64". I am using MacOS Seirra and MacBook Pro

Here is a screenshot attached.
[image]
 
BTW, awesome work guys.

PeiliangLi, 2017-05-17:
build for your device instead of simulator

closed 2017-05-17 by shubh-agrawal

vilin777, 2017-06-10:
I am the same problem.I follow your setting, and project->build, still error. libceres.a is red.
[image]