HKUST-Aerial-Robotics/VINS-Mobile #27: imu-camera extrinsic calibration tools?

zjupx, 2017-05-18:
Really thanks for this open code. 
1,  I want to know which calibration tools you used to calibrate the imu-camera extrinsic for the phone?
do you use kalibr tools for https://github.com/ethz-asl/kalibr/wiki/camera-imu-calibration?
if you use this tool,  I want to know these params for imu noise is right ?
----imu.yaml----
update_rate: 100.0 #Hz
accelerometer_noise_density: 0.5 #continous
accelerometer_random_walk: 0.002 
gyroscope_noise_density: 0.2 #continous
gyroscope_random_walk: 4.0e-5

2,  Another question for coordinate system. As read you code, I think the camera frame and imu/body frame is like this:
      x                                                                                      x
      ^        z                                                                           ^
       |        /                                                                             |
       |      /                                                                               |
       |   /                                                                                  |
Cam|/------------------>y         IMU/Body   y<-------------- |
                                                                                           /
                                                                                        /
                                                                                     /
                                                                                z /

I am right for that ?

PeiliangLi, 2017-05-18:
Hi, @zjupx 
It's not easy for calibrating the extrinsic param accurately using the open source calibration tool because of the temporal offsets between two sensors.
We find the sensor mount positions on the PCB and measure them by hand.

zjupx, 2017-05-18:
thank you for you reply.

the other question has some print error. I upload pic instead:

[image]

am I right for this coordinate system

PeiliangLi, 2017-05-18:
Absolutely right.

closed 2017-05-19 by zjupx

wchen61, 2017-11-16:
Hi, Peiliang. Accroding to zjupx's picture, the camera-imu rotation matrix will rotate 180 with x axis, which matrix will be:
1, 0, 0
0, -1, 0
0, 0, -1
But I know the camera-imu rotation matrix is:
0, -1, 0
1, 0, 0
0, 0, 1
Can you please tell me if I miss something?