HKUST-Aerial-Robotics/VINS-Mobile #33: 为什么我初始化不了

changkaizhao, 2017-05-26:
拿iPhone7P 和 iPhone6（放宽了设备限制）试过 都不能初始化！ 按照提示移动也没有反应！

electricmanlpl, 2017-05-27:
As the authors said, set to release mode。。。

changkaizhao, 2017-05-27:
I did set the **Build COnfiguration** to `Release`  and also uncheck the **Debug executable**

PeiliangLi, 2017-05-27:
hi @changkaizhao please run it on iPhone7 and put your logs and screenshots here.

changkaizhao, 2017-05-27:
the log as below:
```
init ok
init VINS begins
clear state
2017-05-27 10:08:06.093661+0800 VINS_ios[5503:1954220] camera available: YES
2017-05-27 10:08:06.146298+0800 VINS_ios[5503:1954220] Camera: Get device <AVCaptureFigVideoDevice: 0x15bd11b10 [Back Camera][com.apple.avfoundation.avcapturedevice.built-in_video:0]>
2017-05-27 10:08:06.149756+0800 VINS_ios[5503:1954220] Camera: activeFormat <AVCaptureDeviceFormat: 0x174003a20 'vide'/'420v' 1920x1080, { 2- 30 fps}, HRSI:3264x1836, fov:58.040, supports vis, max zoom:95.62 (upscales @1.55), AF System:2, ISO:29.0-464.0, SS:0.000025-0.500000, supports HDR>
2017-05-27 10:08:06.150649+0800 VINS_ios[5503:1954220] Camera: exposure duration (before EV) 0.033333
2017-05-27 10:08:06.196192+0800 VINS_ios[5503:1954220] [Camera] device connected? YES
2017-05-27 10:08:06.196303+0800 VINS_ios[5503:1954220] [Camera] device position back
2017-05-27 10:08:06.211512+0800 VINS_ios[5503:1954220] [Camera] FPS set to 30
2017-05-27 10:08:06.212951+0800 VINS_ios[5503:1954220] layout preview layer
2017-05-27 10:08:06.213010+0800 VINS_ios[5503:1954220] flip bounds
2017-05-27 10:08:06.213098+0800 VINS_ios[5503:1954220] [Camera] created AVCaptureVideoDataOutput
init ok
2017-05-27 10:08:06.973945+0800 VINS_ios[5503:1954298] loop start load voc
Device iPhone6
Device iPhone6s param
system version >= 10.2.1
2017-05-27 10:08:07.182638+0800 VINS_ios[5503:1954220] [LayoutConstraints] Unable to simultaneously satisfy constraints.
	Probably at least one of the constraints in the following list is one you don't want. 
	Try this: 
		(1) look at each constraint and try to figure out which you don't expect; 
		(2) find the code that added the unwanted constraint or constraints and fix it. 
(
    "<_UILayoutSupportConstraint:0x174092340 _UILayoutGuide:0x15bd0fd20.height == 0   (active)>",
    "<_UILayoutSupportConstraint:0x174091fd0 V:|-(0)-[_UILayoutGuide:0x15bd0fd20]   (active, names: '|':UIView:0x15bd06800 )>",
    "<NSLayoutConstraint:0x1740928e0 V:|-(0)-[UIImageView:0x15be11290]   (active, names: '|':UIView:0x15bd06800 )>",
    "<NSLayoutConstraint:0x174092a20 V:[_UILayoutGuide:0x15bd0fd20]-(-20)-[UIImageView:0x15be11290]   (active)>"
)

Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x174092a20 V:[_UILayoutGuide:0x15bd0fd20]-(-20)-[UIImageView:0x15be11290]   (active)>

Make a symbolic breakpoint at UIViewAlertForUnsatisfiableConstraints to catch this in the debugger.
The methods in the UIConstraintBasedLayoutDebugging category on UIView listed in <UIKit/UIView.h> may also be helpful.
2017-05-27 10:08:07.198637+0800 VINS_ios[5503:1954220] deviceOrientationDidChange: 0
2017-05-27 10:08:07.198786+0800 VINS_ios[5503:1954220] deviceOrientationDidChange: 0
2017-05-27 10:08:07.203795+0800 VINS_ios[5503:1954220] deviceOrientationDidChange: 3
2017-05-27 10:08:07.392378+0800 VINS_ios[5503:1954220] timestamp 0.000000
2017-05-27 10:08:07.399310+0800 VINS_ios[5503:1954220] Camera: exposure duration (after EV) 0.009992
imu error 411447.631546 411447.647437 411447.644570
imu error 411447.664589 411447.676827 411447.674599
imu error 411447.694617 411447.706215 411447.704628
imu error 411447.704628 411447.716011 411447.714638
imu error 411447.724648 411447.735604 411447.734658
imu error 411447.734658 411447.745400 411447.744667
imu error 411447.764687 411447.774791 411447.774697
imu error 411447.794693 411447.794379 411447.804726
imu error 411447.824744 411447.823741 411447.834755
TIMER_time_goodfeature: 67.95ms
TIMER_time_feature: 74.56ms
2017-05-27 10:08:07.606992+0800 VINS_ios[5503:1954301] throw img, only should happen at the beginning
imu error 411447.854775 411447.872719 411447.864784
```

PeiliangLi, 2017-05-27:
please run it on **iPhone7** and put your logs and screenshots here.

changkaizhao, 2017-05-27:
here I record the screen [show](https://youtu.be/wJz_r8KLk2g)

PeiliangLi, 2017-05-27:
iPhone 6s and above

changkaizhao, 2017-05-27:
I did switch to ip7p the log is as below:
```
init ok
init VINS begins
clear state
2017-05-27 11:33:45.000482+0800 VINS_ios[2112:1065055] camera available: YES
2017-05-27 11:33:45.010267+0800 VINS_ios[2112:1065055] Camera: Get device <AVCaptureFigVideoDevice: 0x10090f580 [Back Camera][com.apple.avfoundation.avcapturedevice.built-in_video:0]>
2017-05-27 11:33:45.010405+0800 VINS_ios[2112:1065055] Camera: activeFormat <AVCaptureDeviceFormat: 0x170011190 'vide'/'420v' 1920x1080, { 3- 30 fps}, HRSI:4096x2304, fov:59.680, supports vis, max zoom:16.00 (upscales @1.94), AF System:2, ISO:22.0-704.0, SS:0.000005-0.333333>
2017-05-27 11:33:45.010599+0800 VINS_ios[2112:1065055] Camera: exposure duration (before EV) 0.033333
2017-05-27 11:33:45.014338+0800 VINS_ios[2112:1065055] [Camera] device connected? YES
2017-05-27 11:33:45.014368+0800 VINS_ios[2112:1065055] [Camera] device position back
2017-05-27 11:33:45.016970+0800 VINS_ios[2112:1065055] [Camera] FPS set to 30
2017-05-27 11:33:45.017502+0800 VINS_ios[2112:1065055] layout preview layer
2017-05-27 11:33:45.017528+0800 VINS_ios[2112:1065055] flip bounds
2017-05-27 11:33:45.017588+0800 VINS_ios[2112:1065055] [Camera] created AVCaptureVideoDataOutput
init ok
Device iPhone7 plus
Device iPhone7 plus param
system version >= 10.2.1
2017-05-27 11:33:45.661075+0800 VINS_ios[2112:1065129] loop start load voc
2017-05-27 11:33:45.693903+0800 VINS_ios[2112:1065055] [LayoutConstraints] Unable to simultaneously satisfy constraints.
	Probably at least one of the constraints in the following list is one you don't want. 
	Try this: 
		(1) look at each constraint and try to figure out which you don't expect; 
		(2) find the code that added the unwanted constraint or constraints and fix it. 
(
    "<_UILayoutSupportConstraint:0x170085460 _UILayoutGuide:0x100816a20.height == 0   (active)>",
    "<_UILayoutSupportConstraint:0x170081c70 V:|-(0)-[_UILayoutGuide:0x100816a20]   (active, names: '|':UIView:0x1008164a0 )>",
    "<NSLayoutConstraint:0x170083c50 V:|-(0)-[UIImageView:0x100810400]   (active, names: '|':UIView:0x1008164a0 )>",
    "<NSLayoutConstraint:0x170083d90 V:[_UILayoutGuide:0x100816a20]-(-20)-[UIImageView:0x100810400]   (active)>"
)

Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x170083d90 V:[_UILayoutGuide:0x100816a20]-(-20)-[UIImageView:0x100810400]   (active)>

Make a symbolic breakpoint at UIViewAlertForUnsatisfiableConstraints to catch this in the debugger.
The methods in the UIConstraintBasedLayoutDebugging category on UIView listed in <UIKit/UIView.h> may also be helpful.
2017-05-27 11:33:45.698983+0800 VINS_ios[2112:1065055] deviceOrientationDidChange: 0
2017-05-27 11:33:45.699053+0800 VINS_ios[2112:1065055] deviceOrientationDidChange: 0
2017-05-27 11:33:45.699681+0800 VINS_ios[2112:1065055] deviceOrientationDidChange: 3
2017-05-27 11:33:45.718654+0800 VINS_ios[2112:1065055] timestamp 0.000000
2017-05-27 11:33:45.718947+0800 VINS_ios[2112:1065055] Camera: exposure duration (after EV) 0.009801
2017-05-27 11:33:45.723527+0800 VINS_ios[2112:1065055] deviceOrientationDidChange: 6
TIMER_time_goodfeature: 9.48ms
TIMER_time_feature: 9.77ms
2017-05-27 11:33:46.004293+0800 VINS_ios[2112:1065132] throw img, only should happen at the beginning
TIMER_visualize: 2.55ms
TIMER_time_feature: 2.88ms
TIMER_visualize: 1.74ms
TIMER_time_feature: 3.07ms
TIMER_visualize: 1.61ms
TIMER_time_goodfeature: 8.22ms
TIMER_time_feature: 25.08ms
2017-05-27 11:33:46.116697+0800 VINS_ios[2112:1065132] throw img, only should happen at the beginning
TIMER_visualize: 3.49ms
TIMER_time_feature: 2.62ms
TIMER_visualize: 1.58ms
TIMER_time_feature: 3.03ms
TIMER_visualize: 1.58ms
2017-05-27 11:33:46.210139+0800 VINS_ios[2112:1065055] deviceOrientationDidChange: 5
TIMER_time_goodfeature: 8.28ms
TIMER_time_feature: 22.35ms
2017-05-27 11:33:46.213184+0800 VINS_ios[2112:1065132] throw img, only should happen at the beginning
TIMER_visualize: 1.67ms
TIMER_time_feature: 2.86ms
TIMER_visualize: 1.70ms
TIMER_time_feature: 2.96ms
TIMER_visualize: 2.49ms
TIMER_time_goodfeature: 11.04ms
TIMER_time_feature: 18.91ms
2017-05-27 11:33:46.310166+0800 VINS_ios[2112:1065132] throw img, only should happen at the beginning
TIMER_visualize: 1.80ms
TIMER_time_feature: 2.63ms
TIMER_visualize: 1.96ms
TIMER_time_feature: 2.86ms
TIMER_visualize: 1.74ms
TIMER_time_goodfeature: 11.95ms
TIMER_time_feature: 15.80ms
2017-05-27 11:33:46.409024+0800 VINS_ios[2112:1065132] throw img, only should happen at the beginning
TIMER_visualize: 2.25ms
TIMER_time_feature: 3.84ms
TIMER_visualize: 3.36ms
TIMER_time_feature: 3.82ms
TIMER_visualize: 5.21ms
TIMER_time_goodfeature: 10.20ms
TIMER_time_feature: 15.41ms
2017-05-27 11:33:46.512236+0800 VINS_ios[2112:1065132] throw img, only should happen at the beginning
TIMER_visualize: 2.22ms
loop image size width: 480 height: 640
loop vocfile /var/containers/Bundle/Application/961E4197-F63D-43E3-B5A2-0D8FBC47CCBF/VINS_ios.app/brief_k10L6.bin
 loop closure init finish
TIMER_load_voc: 855.81ms
2017-05-27 11:33:46.516949+0800 VINS_ios[2112:1065129] loop load voc finish
TIMER_time_feature: 2.71ms
TIMER_visualize: 1.65ms
TIMER_time_feature: 2.73ms
TIMER_visualize: 1.85ms
TIMER_time_goodfeature: 9.05ms
TIMER_time_feature: 12.38ms
2017-05-27 11:33:46.603854+0800 VINS_ios[2112:1065132] throw img, only should happen at the beginning

```

PeiliangLi, 2017-05-27:
hi，does it always threw Image or only at the begin? If not, please upload the rest log and also the screenshot.
thanks

changkaizhao, 2017-05-27:
the rest of log is 
```
TIMER_time_goodfeature: 14.39ms
TIMER_time_feature: 19.63ms
2017-05-27 11:33:56.022246+0800 VINS_ios[2112:1065132] throw img, only should happen at the beginning
TIMER_visualize: 2.00ms
TIMER_time_feature: 4.44ms
TIMER_visualize: 2.13ms
TIMER_time_feature: 3.96ms
TIMER_visualize: 2.34ms
TIMER_time_goodfeature: 11.09ms
TIMER_time_feature: 15.47ms
2017-05-27 11:33:56.118356+0800 VINS_ios[2112:1065132] wait for imu, only should happen at the beginning
2017-05-27 11:33:56.118446+0800 VINS_ios[2112:1065132] wait for imu, only should happen at the beginning
TIMER_visualize: 2.93ms
2017-05-27 11:33:56.128127+0800 VINS_ios[2112:1065132] wait for imu, only should happen at the beginning
TIMER_time_feature: 3.88ms
2017-05-27 11:33:56.138192+0800 VINS_ios[2112:1065132] wait for imu, only should happen at the beginning
TIMER_visualize: 2.25ms
2017-05-27 11:33:56.148117+0800 VINS_ios[2112:1065132] wait for imu, only should happen at the beginning
2017-05-27 11:33:56.158100+0800 VINS_ios[2112:1065132] wait for imu, only should happen at the beginning
2017-05-27 11:33:56.170599+0800 VINS_ios[2112:1065132] wait for imu, only should happen at the beginning
TIMER_time_feature: 5.69ms
TIMER_visualize: 1.98ms
2017-05-27 11:33:56.178063+0800 VINS_ios[2112:1065132] wait for imu, only should happen at the beginning
2017-05-27 11:33:56.188068+0800 VINS_ios[2112:1065132] wait for imu, only should happen at the beginning
2017-05-27 11:33:56.198034+0800 VINS_ios[2112:1065132] wait for imu, only should happen at the beginning
```

PeiliangLi, 2017-05-30:
hi, please make sure your Device orientation is Landscape Right in General setup.

changkaizhao, 2017-06-05:
@PeiliangLi  Landscape Right switched by default.  :(

hoverpeter, 2017-06-08:
I am having the exact same situation. The system is tested on iPhone 7 but having difficulty to initialize. I can't really use it. Please advise. Thanks so much.

hoverpeter, 2017-06-08:
@changkaizhao @PeiliangLi

hoverpeter, 2017-06-09:
@PeiliangLi Could you help explain how to actually initialize it in terms of the right motion? I thought I got it, but it is not working.

PeiliangLi, 2017-06-09:
@hoverpeter It's not that strictly require the specific motion to initialize successfully.
I guess it's because of some other problems. Have you built it in release version?  If yes, please put more logs and screenshots here so we can debug with it.

hoverpeter, 2017-06-09:
@PeiliangLi hope the following is as informative as you expected. 

2017-06-08 09:06:10.472945-0700 VINS_ios[8942:4274652] [DYMTLInitPlatform] platform initialization successful
init ok
init VINS begins
clear state
2017-06-08 09:06:10.655980-0700 VINS_ios[8942:4274500] [MC] System group container for systemgroup.com.apple.configurationprofiles path is /private/var/containers/Shared/SystemGroup/systemgroup.com.apple.configurationprofiles
2017-06-08 09:06:10.657396-0700 VINS_ios[8942:4274500] [MC] Reading from public effective user settings.
2017-06-08 09:06:10.658528-0700 VINS_ios[8942:4274500] camera available: YES
2017-06-08 09:06:10.686206-0700 VINS_ios[8942:4274500] Camera: Get device <AVCaptureFigVideoDevice: 0x100e19940 [Back Camera][com.apple.avfoundation.avcapturedevice.built-in_video:0]>
2017-06-08 09:06:10.687971-0700 VINS_ios[8942:4274500] Camera: activeFormat <AVCaptureDeviceFormat: 0x174008b30 'vide'/'420v' 1920x1080, { 3- 30 fps}, HRSI:4096x2304, fov:59.680, supports vis, max zoom:16.00 (upscales @1.94), AF System:2, ISO:22.0-704.0, SS:0.000005-0.333333>
2017-06-08 09:06:10.688269-0700 VINS_ios[8942:4274500] Camera: exposure duration (before EV) 0.033333
2017-06-08 09:06:10.700568-0700 VINS_ios[8942:4274500] [Camera] device connected? YES
2017-06-08 09:06:10.700606-0700 VINS_ios[8942:4274500] [Camera] device position back
2017-06-08 09:06:10.704064-0700 VINS_ios[8942:4274500] [Camera] FPS set to 30
2017-06-08 09:06:10.704588-0700 VINS_ios[8942:4274500] layout preview layer
2017-06-08 09:06:10.704612-0700 VINS_ios[8942:4274500] flip bounds
2017-06-08 09:06:10.704662-0700 VINS_ios[8942:4274500] [Camera] created AVCaptureVideoDataOutput
init ok
Device iPhone7
Device iPhone7 param
system version >= 10.2.1
2017-06-08 09:06:11.356855-0700 VINS_ios[8942:4274698] loop start load voc
2017-06-08 09:06:11.412596-0700 VINS_ios[8942:4274500] [LayoutConstraints] Unable to simultaneously satisfy constraints.
	Probably at least one of the constraints in the following list is one you don't want. 
	Try this: 
		(1) look at each constraint and try to figure out which you don't expect; 
		(2) find the code that added the unwanted constraint or constraints and fix it. 
(
    "<_UILayoutSupportConstraint:0x17009e910 _UILayoutGuide:0x100e182b0.height == 0   (active)>",
    "<_UILayoutSupportConstraint:0x17009e280 V:|-(0)-[_UILayoutGuide:0x100e182b0]   (active, names: '|':UIView:0x100e17d30 )>",
    "<NSLayoutConstraint:0x17409ab80 V:|-(0)-[UIImageView:0x100f13cc0]   (active, names: '|':UIView:0x100e17d30 )>",
    "<NSLayoutConstraint:0x17409acc0 V:[_UILayoutGuide:0x100e182b0]-(-20)-[UIImageView:0x100f13cc0]   (active)>"
)

Will attempt to recover by breaking constraint 
<NSLayoutConstraint:0x17409acc0 V:[_UILayoutGuide:0x100e182b0]-(-20)-[UIImageView:0x100f13cc0]   (active)>

Make a symbolic breakpoint at UIViewAlertForUnsatisfiableConstraints to catch this in the debugger.
The methods in the UIConstraintBasedLayoutDebugging category on UIView listed in <UIKit/UIView.h> may also be helpful.
2017-06-08 09:06:11.443549-0700 VINS_ios[8942:4274500] deviceOrientationDidChange: 0
2017-06-08 09:06:11.443598-0700 VINS_ios[8942:4274500] deviceOrientationDidChange: 0
2017-06-08 09:06:11.444718-0700 VINS_ios[8942:4274500] deviceOrientationDidChange: 3
2017-06-08 09:06:11.499487-0700 VINS_ios[8942:4274500] timestamp 0.000000
2017-06-08 09:06:11.499752-0700 VINS_ios[8942:4274500] Camera: exposure duration (after EV) 0.033290
TIMER_time_goodfeature: 80.84ms
TIMER_time_feature: 84.91ms
2017-06-08 09:06:11.819683-0700 VINS_ios[8942:4274702] throw img, only should happen at the beginning
TIMER_visualize: 5.49ms
TIMER_time_feature: 64.01ms
TIMER_visualize: 19.25ms
TIMER_time_feature: 4.84ms
TIMER_visualize: 1.49ms
TIMER_time_goodfeature: 27.45ms
TIMER_time_feature: 33.49ms
2017-06-08 09:06:12.070274-0700 VINS_ios[8942:4274702] throw img, only should happen at the beginning
TIMER_visualize: 6.88ms
TIMER_time_feature: 2.70ms
TIMER_visualize: 2.14ms
TIMER_time_feature: 3.29ms
TIMER_visualize: 1.77ms
TIMER_time_goodfeature: 41.12ms
TIMER_time_feature: 71.56ms
2017-06-08 09:06:12.208767-0700 VINS_ios[8942:4274702] throw img, only should happen at the beginning
TIMER_visualize: 3.67ms
TIMER_time_feature: 2.55ms
TIMER_visualize: 1.38ms
TIMER_time_feature: 3.42ms
TIMER_visualize: 1.71ms
TIMER_time_goodfeature: 14.18ms
[... 92 lines cut ...]
TIMER_visualize: 1.79ms
TIMER_time_feature: 3.42ms
TIMER_visualize: 1.58ms
TIMER_time_feature: 5.42ms
2017-06-08 09:06:13.638429-0700 VINS_ios[8942:4274702] throw img, only should happen at the beginning
TIMER_visualize: 1.79ms
TIMER_time_feature: 3.50ms
TIMER_visualize: 3.24ms
TIMER_time_feature: 3.91ms
TIMER_visualize: 20.77ms
TIMER_time_feature: 9.29ms
2017-06-08 09:06:13.752847-0700 VINS_ios[8942:4274702] throw img, only should happen at the beginning
TIMER_visualize: 4.07ms
TIMER_time_feature: 2.72ms
TIMER_visualize: 1.50ms
TIMER_time_feature: 2.85ms
TIMER_visualize: 1.92ms
TIMER_time_feature: 8.35ms
2017-06-08 09:06:13.842891-0700 VINS_ios[8942:4274702] throw img, only should happen at the beginning
TIMER_visualize: 2.50ms
TIMER_time_feature: 13.33ms
TIMER_visualize: 1.53ms
TIMER_time_feature: 7.84ms
TIMER_visualize: 1.78ms
TIMER_time_feature: 33.43ms

PeiliangLi, 2017-06-09:
@hoverpeter It seem the same error with @changkaizhao caused by some weird iOS UI constraints problem. You can refer https://developer.apple.com/library/content/documentation/UserExperience/Conceptual/AutolayoutPG/DebuggingTricksandTips.html

hoverpeter, 2017-06-09:
@PeiliangLi I'll give it a try, thanks for the tips. Will let u know if it can work out.

hoverpeter, 2017-06-09:
@PeiliangLi I don't think UI constraints played a major to this problem. I also get message like below

"wait for imu, only should happen at the beginning"

hoverpeter, 2017-06-13:
Problem solved on my side. 

In the file ViewController.mm, line 333 should be commented out. Not sure whey it is left uncommented there. It definitely caused the wrong value assignment for img_msg.

PeiliangLi, 2017-06-13:
@hoverpeter so you didn't use the provided opencv2.framework or have installed opencv in your OSX?
Because the timestamp is packed in the first two pixels of the image in the provided opencv

hoverpeter, 2017-06-13:
@PeiliangLi my bad, I just realized that. Yes, now it is working all perfectly. Could you explain what was the reason to modify the original OpenCV?

PeiliangLi, 2017-06-14:
@changkaizhao Because opencv for iOS doesn't provide image timestamp API, so we need to 'hack'  it to get the timestamp.

changkaizhao, 2017-06-15:
@PeiliangLi  thanks it solved the problem!

closed 2017-06-18 by PeiliangLi

xhniu, 2021-08-10:
> @PeiliangLi thanks it solves the problem! my fault, I use the OpenCV downloaded from official website!

Hi, where to download specifical opencv2.framework? The original link provided had been invalid!