HKUST-Aerial-Robotics/VINS-Mobile #37: how to calc/set/adjust the params for new device ,eg iphone6p

tigerking, 2017-06-02:
I do some modification to support iphone6p

1. file: global_param.hpp
enum DeviceType
{
    iPhone7P,
    iPhone7,
    iPhone6sP,
    iPhone6s,
// +++++
    iPhone6,
    iPhone6P,
// +++++++ END
    iPadPro97,
    iPadPro129,
    unDefine
};

2. file: ViewController.mm
DeviceType deviceName()
{
...
// ++++ 
    else if(([device compare:@"iPhone7,1"] == NSOrderedSame))
    {
        printf("Device iPhone6p\n");
        device_type = iPhone6P;
    }
//+++ END
}

3. file:  global_param.cpp
bool setGlobalParam(DeviceType device)
{
//+++++ , params copied from  iphone6sp

      case iPhone6P:
            printf("Device iPhone6P param\n");
            FOCUS_LENGTH_X = 547.565;
            FOCUS_LENGTH_Y = 547.998;
            PX = 239.033;
            PY = 309.452;
            
            SOLVER_TIME = 0.08;
            FREQ = 3;
            
            //extrinsic param
            TIC_X = 0.0;
            TIC_Y = 0.065;
            TIC_Z = 0.0;
            return true;
            break;
// ++++ END

PeiliangLi, 2017-06-02:
It's also because iPhone6 is not enough to solve it in real time.

tigerking, 2017-06-02:
if these parameters are also one fact that affect the result ?. thx

PeiliangLi, 2017-06-02:
Yes, right. You can tune these params for better performance.

closed 2017-06-14 by PeiliangLi