HKUST-Aerial-Robotics/VINS-Mobile #41: Time jump for image?

guohengkai, 2017-06-06:
In ViewController.mm --> process(), when it fail to obtain newest processed image in image_pool and vins_pool, the function will use original image. Does it cause the time jump on the image because the original image timestamp is after that of the processed image?

PeiliangLi, 2017-06-06:
if it fails to obtain the newest image, the measurement will be empty, it won't enter vins.processImage()

guohengkai, 2017-06-06:
Sorry, maybe I have not expressed clearly. In the function processImage(), the "image" is first used to extract features. Then from L459-486, the program tries to obtain processed image from image_pool and replace the "image" with it. And finally the "image" will be shown on the view. 

When failed to do so, the "image" will be the original one, whose timestamp is newer than that in the image_pool. So there maybe a jump on timestamp, right?

PeiliangLi, 2017-06-07:
The replace is only for AR visualization, because we need show the image where the pose is solved by vins.
The image will never be the origin one, because the image_pool is always non-empty.
[image]

guohengkai, 2017-06-08:
Thanks for your response. The image_pool is indeed always non-empty, but the vins_pool may be. This is what the problem is.

PeiliangLi, 2017-06-08:
vins_poll cannot be empty, only pop it when the size > 1
[image]

guohengkai, 2017-06-08:
See `if(!vins_pool.empty() && !image_pool.empty())`. Only when they are not empty, the codes you showed will be executed. Or it will skip them.

PeiliangLi, 2017-06-08:
it only be empty in initialization process

closed 2017-06-14 by PeiliangLi