HKUST-Aerial-Robotics/VINS-Mobile #43: jacobian_pose_i.maxCoeff and minCoeff is out of range

FangGet, 2017-06-08:
After initialized(with s equal to 0.07?is this normal?), program assert failed at imu_factor.h for both jacobian_pose_i.maxCoeff and jacobian_pose_i.minCoeff out of range( e^22), can you tell what would cause this?

PeiliangLi, 2017-06-09:
s might be arbitrary because it represents the scale of the vision-only translation.
Assert false may occur when you keep your device static in a long time, the imu covariance will become extremely large. We assert this to avoid the numerical issue.

FangGet, 2017-06-09:
Since you system init with optical flow, I'm pretty sure I initialize it with a pure translation, its device static the only factor to the result?
btw, it's a migrate android version.

PeiliangLi, 2017-06-09:
Have you ported it onto Android device？

FangGet, 2017-06-09:
yeap, but extrinsic param is not calculated.

PeiliangLi, 2017-06-09:
It can not directly run on Android devices unless you calibrate your camera intrinsic and set a right rotation and approximate translation params.

FangGet, 2017-06-09:
intrinsic param is obtained while extrinsic remains to do. coordinate system seems to be the same as iphone, so I'm trying to set the proper param for translation. how much does translation influence the jacobian elements?

PeiliangLi, 2017-06-14:
@FangGet translation won't affect the results significantly as long as you use the right intrinsic and rotation params.
We cannot determine what reasons cause your problem. Because porting the whole pipeline to another platform is not a straightforward work, there are too many details need to be considered

ngupta05, 2017-06-22:
Hi @FangGet, we are also trying to port it to android, but our code gets stuck at initialization currently. We didn't find the coordinate system to be same for iphone & android, so trying to correct that along with other params. Were you able to make progress on it? If so, could you please let us know if there are any specific things we need to keep in mind while porting.

juwikuang, 2017-06-22:
count me in. I introduced VINS to my boss. now my boss says iPhone has arkit, we do not need VINS on iPhone. We need VINS on Android phones. Tell me if any of you has done it successfully. I will start working on it soon and share my work here, too. @FangGet @ngupta05

TouqeerAhmad, 2017-08-24:
@FangGet Hi, would you be able to share the android version of VINS ???