HKUST-Aerial-Robotics/VINS-Mobile #44: a problem about scale initialization

vonzy, 2017-06-11:
I read your paper and use your method for the initialization. I can get the right gyroscope bias, but can't get the correct scale, and it's very unstable. Here is the result for scale.
[image]

I don't know what's wrong with it. The equation I used is as follow:
[image]

qintony, 2017-06-12:
In this implementation code, we relax the initialization threshold for the better experience.

It's just an initial value to bootstrap the nonlinear system. We don't use it to calibrate the real scale all the time. 

In the paper, we use the strict threshold. Such as strict g limitation, IMU variance, and SfM cost.

vonzy, 2017-06-12:
@qintony so if the threshold is relaxed, the scale could get wrong possibly (even negative? I wonder if my code get wrong?). could you tell me the strict threshold you used? thanks!!!

vonzy, 2017-06-12:
@qintony moreover, the condition number of H is very huge, so the equation is easily influenced by any errors in b?

qintony, 2017-06-12:
The equation is right. 
If alpha and beta don't contain much excitation (such as near stable, zero acceleration), the equation is fully unobservable, the H will be ill-conditioned, and any result will be solved.  Sometimes negatives happen.  The preconditions are that, alpha, beta and SfM structure should be as accurate as possible.  

Following are thresholds

accurate sfm
[image]

good gravity
[image]

enough IMU excitation
[image]

As for a general application, nobody wants to wait a long time to get a much accurate initial value. So we relax these thresholds, get a rough initial guess (much better than no initial guess).

vonzy, 2017-06-13:
all right! thanks!

closed 2017-06-13 by vonzy