HKUST-Aerial-Robotics/VINS-Mobile #45: Build Framework failed

ChokWah, 2017-06-12:
I want to  build  VINS-ios-framework Target  to  a framework for my project , build failed.
// step
Selected  VINS-ios-framework Target
add VINS_ios - info.plist to  VINS-ios-framework Target's info
Build Failed  : 
[image]

Maybe I did something wrong . Could you tell me about this ,Please?

ChokWah, 2017-06-12:
Oh, My mistake.
I made some change in buildSetting like this . It works : Build Succeeded.

[image]

closed 2017-06-14 by PeiliangLi

Varvrar, 2019-04-10:
Hello ChokWah!
I manage to build VINS-ios-framework by adding missing .cpp files to Compile Sources. But I still can't understand how to use this framework in my app. It does not even have public headers.

Did you manage to use this framework in your project?

yangxu-debug, 2025-06-26:
[image] can anyone help me in this problem when building project?