HKUST-Aerial-Robotics/VINS-Mobile #48: confused about c calculation in the code

vonzy, 2017-06-15:
the matrix V in your paper is  a 15x12 matrix .
[image]
but why in the code it's a 15x18 matrix ?
` MatrixXd V = MatrixXd::Zero(15,18);
            V.block<3, 3>(0, 0) =  0.25 * delta_q.toRotationMatrix() * _dt * _dt;
            V.block<3, 3>(0, 3) =  0.25 * -result_delta_q.toRotationMatrix() * R_a_1_x  * _dt * _dt * 0.5 * _dt;
            V.block<3, 3>(0, 6) =  0.25 * result_delta_q.toRotationMatrix() * _dt * _dt;
            V.block<3, 3>(0, 9) =  V.block<3, 3>(0, 3);
            V.block<3, 3>(3, 3) =  0.5 * MatrixXd::Identity(3,3) * _dt;
            V.block<3, 3>(3, 9) =  0.5 * MatrixXd::Identity(3,3) * _dt;
            V.block<3, 3>(6, 0) =  0.5 * delta_q.toRotationMatrix() * _dt;
            V.block<3, 3>(6, 3) =  0.5 * -result_delta_q.toRotationMatrix() * R_a_1_x  * _dt * 0.5 * _dt;
            V.block<3, 3>(6, 6) =  0.5 * result_delta_q.toRotationMatrix() * _dt;
            V.block<3, 3>(6, 9) =  V.block<3, 3>(6, 3);
            V.block<3, 3>(9, 12) = MatrixXd::Identity(3,3) * _dt;
            V.block<3, 3>(12, 15) = MatrixXd::Identity(3,3) * _dt;`

PeiliangLi, 2017-06-16:
The paper uses Euler integration to simplify representation. 
However, In the code we use mid-point integration for better accuracy. So the noise covariance matrix has 18 dimensions(2 acc_N, 2 gyr_N, 1 acc_W, 1 gyr_W), and the V(denoted as G in the paper) is 15x18 matrix.

vonzy, 2017-06-17:
@PeiliangLi get it! thanks!

closed 2017-06-18 by vonzy