HKUST-Aerial-Robotics/VINS-Mobile #49: clang: error: linker command failed with exit code 1 (use -v to see invocation)

hunkyu, 2017-06-15:
Ld /Users/macbook-liang/Library/Developer/Xcode/DerivedData/VINS_ios-aoxynvrebnnkzzbqjqbjamczosua/Build/Products/Release-iphonesimulator/VINS_ios.app/VINS_ios normal x86_64
    cd /Users/macbook-liang/Downloads/VINS-Mobile-master
    export IPHONEOS_DEPLOYMENT_TARGET=10.0
    export PATH="/Users/macbook-liang/Downloads/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/usr/bin:/Users/macbook-liang/Downloads/Xcode.app/Contents/Developer/usr/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin"
    /Users/macbook-liang/Downloads/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -arch x86_64 -isysroot /Users/macbook-liang/Downloads/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator10.2.sdk -L/Users/macbook-liang/Library/Developer/Xcode/DerivedData/VINS_ios-aoxynvrebnnkzzbqjqbjamczosua/Build/Products/Release-iphonesimulator -L/Users/macbook-liang/Downloads/VINS-Mobile-master/VINS_ThirdPartyLib/ceres-solver/ceres-bin/lib -L/Users/macbook-liang/Downloads/VINS-Mobile-master/ThirdParty -L/Users/macbook-liang/Downloads/VINS-Mobile-master/Resources -F/Users/macbook-liang/Library/Developer/Xcode/DerivedData/VINS_ios-aoxynvrebnnkzzbqjqbjamczosua/Build/Products/Release-iphonesimulator -F/Users/macbook-liang/Downloads/VINS-Mobile-master/VINS_ThirdPartyLib -F/Users/macbook-liang/Downloads/VINS-Mobile-master/ThirdParty -filelist /Users/macbook-liang/Library/Developer/Xcode/DerivedData/VINS_ios-aoxynvrebnnkzzbqjqbjamczosua/Build/Intermediates/VINS_ios.build/Release-iphonesimulator/VINS_ios.build/Objects-normal/x86_64/VINS_ios.LinkFileList -Xlinker -rpath -Xlinker @executable_path/Frameworks -mios-simulator-version-min=10.0 -dead_strip -Xlinker -object_path_lto -Xlinker /Users/macbook-liang/Library/Developer/Xcode/DerivedData/VINS_ios-aoxynvrebnnkzzbqjqbjamczosua/Build/Intermediates/VINS_ios.build/Release-iphonesimulator/VINS_ios.build/Objects-normal/x86_64/VINS_ios_lto.o -Xlinker -objc_abi_version -Xlinker 2 -stdlib=libc++ -fobjc-arc -fobjc-link-runtime -Xlinker -sectcreate -Xlinker __TEXT -Xlinker __entitlements -Xlinker /Users/macbook-liang/Library/Developer/Xcode/DerivedData/VINS_ios-aoxynvrebnnkzzbqjqbjamczosua/Build/Intermediates/VINS_ios.build/Release-iphonesimulator/VINS_ios.build/VINS_ios.app.xcent /Users/macbook-liang/Downloads/VINS-Mobile-master/Resources/boost.a -ljpeg -framework opencv2 -framework CoreMotion -framework GLKit -framework CoreGraphics -framework OpenGLES -lceres -framework Foundation -framework UIKit -framework AVFoundation -framework AssetsLibrary -framework Accelerate -framework CoreVideo -framework CoreMedia -framework CoreImage -framework QuartzCore -Xlinker -dependency_info -Xlinker /Users/macbook-liang/Library/Developer/Xcode/DerivedData/VINS_ios-aoxynvrebnnkzzbqjqbjamczosua/Build/Intermediates/VINS_ios.build/Release-iphonesimulator/VINS_ios.build/Objects-normal/x86_64/VINS_ios_dependency_info.dat -o /Users/macbook-liang/Library/Developer/Xcode/DerivedData/VINS_ios-aoxynvrebnnkzzbqjqbjamczosua/Build/Products/Release-iphonesimulator/VINS_ios.app/VINS_ios

ld: library not found for -lceres
clang: error: linker command failed with exit code 1 (use -v to see invocation)

hunkyu, 2017-06-15:
Ceres is not to recompile the build library file? Originally libceres.a found a lot of link error, then I changed to SHARED, the above mistakes

PeiliangLi, 2017-06-16:
Please upload your screenshots after building it.

hunkyu, 2017-06-16:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(canonical_views_clustering.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(cxsparse.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(single_linkage_clustering.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(suitesparse.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(visibility.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(visibility_based_preconditioner.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(canonical_views_clustering.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(cxsparse.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(single_linkage_clustering.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(suitesparse.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(visibility.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(visibility_based_preconditioner.cc.o) has no symbols

hunkyu, 2017-06-16:
VINS_ios Group
Documentation Issue Group
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core/matx.hpp:187:19: Not a Doxygen trailing comment
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:57:10: In file included from /Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:57:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core/types.hpp:86:17: Not a Doxygen trailing comment
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:10: In file included from /Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core/types.hpp:175:15: Not a Doxygen trailing comment
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:10: In file included from /Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core/types.hpp:243:18: Not a Doxygen trailing comment
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:10: In file included from /Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core/types.hpp:403:30: Not a Doxygen trailing comment
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:10: In file included from /Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core/types.hpp:482:21: Not a Doxygen trailing comment
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:10: In file included from /Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core/types.hpp:483:21: Not a Doxygen trailing comment
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:10: In file included from /Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core/types.hpp:484:21: Not a Doxygen trailing comment
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/ScoringObject.cpp:11:
/Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:10: In file included from /Users/macbook-liang/VINS-Mobile/ThirdParty/DBoW/TemplatedVocabulary.h:20:
/Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:10: In file included from /Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/opencv2.framework/Headers/core.hpp:58:
[... cut ...]
  "ceres::Solver::Summary::Summary()", referenced from:

  "ceres::Solve(ceres::Solver::Options const&, ceres::Problem*, ceres::Solver::Summary*)", referenced from:

  "vtable for ceres::QuaternionParameterization", referenced from:

  "typeinfo for ceres::LocalParameterization", referenced from:

  "ceres::Problem::RemoveResidualBlock(ceres::internal::ResidualBlock*)", referenced from:

  "ceres::Problem::~Problem()", referenced from:

clang: error: linker command failed with exit code 1 (use -v to see invocation)

hunkyu, 2017-06-16:
Here is my re-build ceres-package

hunkyu, 2017-06-16:
===============================================================
-- No preference for use of exported gflags CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported gflags CMake configuration if available.
-- Found installed version of gflags: /usr/local/lib/cmake/gflags
-- Detected gflags version: 2.2.0
-- Found Gflags: /usr/local/include  
-- Found Google Flags header in: /usr/local/include, in namespace: google
-- No preference for use of exported glog CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported glog CMake configuration if available.
-- Found installed version of glog: /usr/local/lib/cmake/glog
-- Detected glog version: 0.3.5
-- Found Glog: glog::glog  
-- Found Google Log (glog). Assuming glog was built with gflags support as gflags was found. This will make gflags a public dependency of Ceres.
-- Failed to find OpenMP, disabling. This is expected on Clang < 3.8, and at least Xcode <= 7.
-- Performing Test COMPILER_HAS_CXX11_FLAG
-- Performing Test COMPILER_HAS_CXX11_FLAG - Success
-- Looking for C++ include unordered_map
-- Looking for C++ include unordered_map - found
-- Performing Test HAVE_UNORDERED_MAP_IN_STD_NAMESPACE
-- Performing Test HAVE_UNORDERED_MAP_IN_STD_NAMESPACE - Success
-- Found unordered_map/set in std namespace.
-- Looking for C++ include memory
-- Looking for C++ include memory - found
-- Performing Test HAVE_SHARED_PTR_IN_STD_NAMESPACE
-- Performing Test HAVE_SHARED_PTR_IN_STD_NAMESPACE - Success
-- Found shared_ptr in std namespace using <memory> header.
-- Building Ceres as a static library.
-- No build type specified; defaulting to CMAKE_BUILD_TYPE=Release.
-- Performing Test HAVE_RETURN_TYPE_C_LINKAGE
-- Performing Test HAVE_RETURN_TYPE_C_LINKAGE - Success
-- Creating configured Ceres config.h output directory: /Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/ceres-solver/build/config/ceres/internal
-- Enabling CERES_NO_SUITESPARSE in Ceres config.h
-- Enabling CERES_NO_CXSPARSE in Ceres config.h
-- Enabling CERES_NO_THREADS in Ceres config.h
-- Enabling CERES_STD_UNORDERED_MAP in Ceres config.h
-- Build the examples.
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/ceres-solver/build

hunkyu, 2017-06-16:
Some errors after building 。
Linking CXX static library ../../lib/libceres.a
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(canonical_views_clustering.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(cxsparse.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(single_linkage_clustering.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(suitesparse.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(visibility.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(visibility_based_preconditioner.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(canonical_views_clustering.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(cxsparse.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(single_linkage_clustering.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(suitesparse.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(visibility.cc.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: ../../lib/libceres.a(visibility_based_preconditioner.cc.o) has no symbols
[ 58%] Built target ceres

hunkyu, 2017-06-16:
.............
[ 97%] Building CXX object examples/CMakeFiles/robot_pose_mle.dir/robot_pose_mle.cc.o
Linking CXX executable ../bin/robot_pose_mle
[ 97%] Built target robot_pose_mle
Scanning dependencies of target robust_curve_fitting
[ 97%] Building CXX object examples/CMakeFiles/robust_curve_fitting.dir/robust_curve_fitting.cc.o
Linking CXX executable ../bin/robust_curve_fitting
[ 97%] Built target robust_curve_fitting
Scanning dependencies of target rosenbrock
[ 98%] Building CXX object examples/CMakeFiles/rosenbrock.dir/rosenbrock.cc.o
Linking CXX executable ../bin/rosenbrock
[ 98%] Built target rosenbrock
Scanning dependencies of target simple_bundle_adjuster
[ 98%] Building CXX object examples/CMakeFiles/simple_bundle_adjuster.dir/simple_bundle_adjuster.cc.o
Linking CXX executable ../bin/simple_bundle_adjuster
[ 98%] Built target simple_bundle_adjuster
Scanning dependencies of target sampled_function
[ 99%] Building CXX object examples/sampled_function/CMakeFiles/sampled_function.dir/sampled_function.cc.o
Linking CXX executable ../../bin/sampled_function
[ 99%] Built target sampled_function
Scanning dependencies of target pose_graph_2d
[ 99%] Building CXX object examples/slam/pose_graph_2d/CMakeFiles/pose_graph_2d.dir/pose_graph_2d.cc.o
Linking CXX executable ../../../bin/pose_graph_2d
[ 99%] Built target pose_graph_2d
Scanning dependencies of target pose_graph_3d
[100%] Building CXX object examples/slam/pose_graph_3d/CMakeFiles/pose_graph_3d.dir/pose_graph_3d.cc.o
Linking CXX executable ../../../bin/pose_graph_3d

1. I put the directory ..../VINS_ThirdPartyLib/ceres-solver/build/lib/libceres.a  under cmake-bin
2. xcode rebuild is failed

hunkyu, 2017-06-18:
Ld /Users/liangliangmac/Library/Developer/Xcode/DerivedData/VINS_ios-dirivyfoxqwsukecregvyddcynsa/Build/Products/Release-iphonesimulator/VINS_ios.app/VINS_ios normal x86_64
    cd /Users/liangliangmac/Downloads/VINS-Mobile-master
    export IPHONEOS_DEPLOYMENT_TARGET=8.0
    export PATH="/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/usr/bin:/Applications/Xcode.app/Contents/Developer/usr/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin"
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -arch x86_64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator10.3.sdk -L/Users/liangliangmac/Library/Developer/Xcode/DerivedData/VINS_ios-dirivyfoxqwsukecregvyddcynsa/Build/Products/Release-iphonesimulator -L/Users/liangliangmac/Downloads/VINS-Mobile-master/VINS_ThirdPartyLib/ceres-solver/ceres-bin/lib -L/Users/liangliangmac/Downloads/VINS-Mobile-master/ThirdParty -L/Users/liangliangmac/Downloads/VINS-Mobile-master/Resources -F/Users/liangliangmac/Library/Developer/Xcode/DerivedData/VINS_ios-dirivyfoxqwsukecregvyddcynsa/Build/Products/Release-iphonesimulator -F/Users/liangliangmac/Downloads/VINS-Mobile-master/VINS_ThirdPartyLib -F/Users/liangliangmac/Downloads/VINS-Mobile-master/ThirdParty -filelist /Users/liangliangmac/Library/Developer/Xcode/DerivedData/VINS_ios-dirivyfoxqwsukecregvyddcynsa/Build/Intermediates/VINS_ios.build/Release-iphonesimulator/VINS_ios.build/Objects-normal/x86_64/VINS_ios.LinkFileList -Xlinker -rpath -Xlinker @executable_path/Frameworks -mios-simulator-version-min=8.0 -dead_strip -Xlinker -object_path_lto -Xlinker /Users/liangliangmac/Library/Developer/Xcode/DerivedData/VINS_ios-dirivyfoxqwsukecregvyddcynsa/Build/Intermediates/VINS_ios.build/Release-iphonesimulator/VINS_ios.build/Objects-normal/x86_64/VINS_ios_lto.o -Xlinker -objc_abi_version -Xlinker 2 -stdlib=libc++ -fobjc-arc -fobjc-link-runtime -Xlinker -sectcreate -Xlinker __TEXT -Xlinker __entitlements -Xlinker /Users/liangliangmac/Library/Developer/Xcode/DerivedData/VINS_ios-dirivyfoxqwsukecregvyddcynsa/Build/Intermediates/VINS_ios.build/Release-iphonesimulator/VINS_ios.build/VINS_ios.app.xcent /Users/liangliangmac/Downloads/VINS-Mobile-master/Resources/boost.a -ljpeg -lglog -framework opencv2 -framework CoreMotion -framework GLKit -framework CoreGraphics -framework OpenGLES -lceres -framework Foundation -framework UIKit -framework AVFoundation -framework AssetsLibrary -framework Accelerate -framework CoreVideo -framework CoreMedia -framework CoreImage -framework QuartzCore -Xlinker -dependency_info -Xlinker /Users/liangliangmac/Library/Developer/Xcode/DerivedData/VINS_ios-dirivyfoxqwsukecregvyddcynsa/Build/Intermediates/VINS_ios.build/Release-iphonesimulator/VINS_ios.build/Objects-normal/x86_64/VINS_ios_dependency_info.dat -o /Users/liangliangmac/Library/Developer/Xcode/DerivedData/VINS_ios-dirivyfoxqwsukecregvyddcynsa/Build/Products/Release-iphonesimulator/VINS_ios.app/VINS_ios

ld: library not found for -lglog
clang: error: linker command failed with exit code 1 (use -v to see invocation)

ageofish, 2017-07-04:
@hunkyu you probably have already found out why. I run into a similar issue:
libceres.a is built for arm only, no x86_64. So don't build and run it on simulator.

check this warning in your log: ignoring file /Users/macbook-liang/VINS-Mobile/VINS_ThirdPartyLib/ceres-solver/ceres-bin/lib/libceres.a, file was built for archive which is not the architecture being linked

closed 2017-07-17 by PeiliangLi