HKUST-Aerial-Robotics/VINS-Mobile #50: about run holokit

yayachu, 2017-06-16:
I follow the documentation steps you have given to run holokit, but the following problems, a moment can not be resolved, please help solve: 
1): "vtable for ceres::HuberLoss", referenced from:
2): "ceres::Problem::AddResidualBlock(ceres::CostFunction*, ceres::LossFunction*, double*, double*, double*)", referenced from:
3): "ceres::Solver::Summary::BriefReport() const", referenced from:
4): "ceres::LocalParameterization::~LocalParameterization()", referenced from:
5): "ceres::Problem::SetParameterBlockConstant(double*)", referenced from:
6): "google::log_sinks_global", referenced from:
7): "vtable for ceres::CauchyLoss", referenced from:
8): "ceres::LocalParameterization::MultiplyByJacobian(double const*, int, double const*, double*) const", referenced from:
9): "ceres::Problem::AddResidualBlock(ceres::CostFunction*, ceres::LossFunction*, std::__1::vector<double*, std::__1::allocator<double*> > const&)", referenced from:
10): "ceres::Problem::AddParameterBlock(double*, int)", referenced from:
11): "ceres::Problem::AddParameterBlock(double*, int, ceres::LocalParameterization*)", referenced from:
12): "ceres::Problem::Problem()", referenced from:
13): "ceres::Problem::GetResidualBlocks(std::__1::vector<ceres::internal::ResidualBlock*, std::__1::allocator<ceres::internal::ResidualBlock*> >*) const", referenced from:
14): "ceres::Problem::AddResidualBlock(ceres::CostFunction*, ceres::LossFunction*, double*, double*, double*, double*)", referenced from:
15): "ceres::Solver::Summary::Summary()", referenced from:
16): "ceres::Solve(ceres::Solver::Options const&, ceres::Problem*, ceres::Solver::Summary*)", referenced from:
17): "vtable for ceres::QuaternionParameterization", referenced from:
18): "typeinfo for ceres::LocalParameterization", referenced from:
19): "ceres::Problem::RemoveResidualBlock(ceres::internal::ResidualBlock*)", referenced from:
20): "ceres::Problem::~Problem()", referenced from:
21): Linker command failed with exit code 1 (use -v to see invocation)

Thank you!

PetWorm, 2017-06-16:
might help u
u should add the team and use a unique bundle identifier, try add some random string after the given one

发自我的 iPhone

PeiliangLi, 2017-06-16:
Please do not build for the simulator.

closed 2017-07-17 by PeiliangLi