HKUST-Aerial-Robotics/VINS-Mobile #52: very confused about jacobians

vonzy, 2017-06-17:
[image]
2. very very confused about imu factor part in the code!!  I read the paper of Forster 'IMU Preintegration on Manifold for Efficient Visual-Inertial Maximum-a-Posteriori Estimation', and find there are some difference about residual jocabians between you used and Forster  provided, can you provide the residual jocabians equations you used? 
3. Sorry for no developing experience with Eigen and Ceres yet, a very simple question maybe:  is function 'virtual bool Evaluate(double const *const *parameters, double *residuals, double **jacobians)' in ceres solver used to calculate residuals with the jacobians provided by developer? If so, I can't find where the code assign value to jacobians. I can only find the code assign value to matrixs like jacobian_pose_i,jacobian_pose_j and etc.

vonzy, 2017-06-21:
Is anybody avaliable?

vonzy, 2017-06-21:
@PeiliangLi

PeiliangLi, 2017-06-21:
1.  NO. 
[image]
2. Check our technical report https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/support_files/paper/tro_technical_report.pdf and http://ieeexplore.ieee.org/document/7139939/ carefully, then you can find its derivation.
3. Yes. The jacobian is assigned here:
[image]

vonzy, 2017-06-21:
get it,thanks!

closed 2017-07-17 by PeiliangLi