HKUST-Aerial-Robotics/VINS-Mobile #53: opencv not found and use issues

kuonangzhe, 2017-06-17:
Hi, 

I Build and Run the code while it said the problem: 
"
ld: framework not found opencv2
clang: error: linker command failed with exit code 1 (use -v to see invocation)
"
At the begining in the opencv2.framework, since the path is weird with only four stuff: headers, version, resource and unaccessible file named 'opencv2'. while the opencv file path used in project was like opencv/core/..., so I changed all into opencv/hearders/core/.... Is this related with the problem? But what i did solved those 'file not found' problem.

Can anyone help me with this? Thanks a lot!

cjanietz, 2017-06-17:
You are following the instructions? OpenCV is a separate download which you have to extract

kuonangzhe, 2017-06-17:
Yeah I did. I first uninstalled opencv in my labtop and then downloaded the opencv2.framework from the provided link. While the extracted opencv folder contains four parts as I mentioned, but not the usual opencv path structure.

PeiliangLi, 2017-06-17:
@kuonangzhe The opencv2.framwork is for iOS, so it doesn't look like the usual version.
Which version Xcode are you using and provide **screenshots** after you build it.

kuonangzhe, 2017-06-18:
[image]
[image]
Xcode is in version 8.2.1

PeiliangLi, 2017-06-19:
hi, You don't need to change anything except your own bundle id.
Just follow the README.