HKUST-Aerial-Robotics/VINS-Mobile #54: help ,thanks

sdkmsdn, 2017-06-17:
I follow the instructions to download the codes and opencv2.framework  development package, but the compiler can not pass.
my os : macOS 10.12.5
xcode: 8.3.3

error information is in following :
Apple Mach-O Linker Error Group
  "vtable for ceres::HuberLoss", referenced from:

  "ceres::Problem::AddResidualBlock(ceres::CostFunction*, ceres::LossFunction*, double*, double*, double*)", referenced from:

  "ceres::Solver::Summary::BriefReport() const", referenced from:

  "ceres::LocalParameterization::~LocalParameterization()", referenced from:

  "ceres::Problem::SetParameterBlockConstant(double*)", referenced from:

  "google::log_sinks_global", referenced from:

  "vtable for ceres::CauchyLoss", referenced from:

  "ceres::LocalParameterization::MultiplyByJacobian(double const*, int, double const*, double*) const", referenced from:

  "ceres::Problem::AddResidualBlock(ceres::CostFunction*, ceres::LossFunction*, std::__1::vector<double*, std::__1::allocator<double*> > const&)", referenced from:

  "ceres::Problem::AddParameterBlock(double*, int)", referenced from:

  "ceres::Problem::AddParameterBlock(double*, int, ceres::LocalParameterization*)", referenced from:

  "ceres::Problem::Problem()", referenced from:

  "ceres::Problem::GetResidualBlocks(std::__1::vector<ceres::internal::ResidualBlock*, std::__1::allocator<ceres::internal::ResidualBlock*> >*) const", referenced from:

  "ceres::Problem::AddResidualBlock(ceres::CostFunction*, ceres::LossFunction*, double*, double*, double*, double*)", referenced from:

  "ceres::Solver::Summary::Summary()", referenced from:

  "ceres::Solve(ceres::Solver::Options const&, ceres::Problem*, ceres::Solver::Summary*)", referenced from:

  "vtable for ceres::QuaternionParameterization", referenced from:

  "typeinfo for ceres::LocalParameterization", referenced from:

  "ceres::Problem::RemoveResidualBlock(ceres::internal::ResidualBlock*)", referenced from:

  "ceres::Problem::~Problem()", referenced from:

clang: error: linker command failed with exit code 1 (use -v to see invocation)

hunkyu, 2017-06-18:
I also encountered the same problem, I feel should be ceres-solver the cmake-bin file need to re-compile their own libceres.a

sdkmsdn, 2017-06-18:
Please tell me the detailed steps, thank you!

PeiliangLi, 2017-06-19:
libceres.a is included in the repo, you don't need to compile it.
https://github.com/HKUST-Aerial-Robotics/VINS-Mobile/blob/master/VINS_ThirdPartyLib/ceres-solver/ceres-bin/lib/libceres.a
Until now, all the ceres problem is because of selecting simulator as build target.

sdkmsdn, 2017-06-22:
I have been running on ipad2 above, and really a great program, thank you very much for your open source!
In addition: I go a circle, but the program trajectory is not closed, which is caused by mems data drift? Or other factors? Thank you!

PeiliangLi, 2017-06-22:
To detect loop, the camera needs to revisit the similar characteristic scene.

closed 2017-07-17 by PeiliangLi