HKUST-Aerial-Robotics/VINS-Mobile #55: Another question about imu factor

vonzy, 2017-06-18:
[image]

PeiliangLi, 2017-06-19:
1. q has only 3 DoF
2. It's not solved by close form. It was optimized by non-linear, we provide a good initial guess, and the cost function will converge in the minima.

vonzy, 2017-06-19:
[image]
@PeiliangLi

vonzy, 2017-06-19:
[image]
@PeiliangLi

PeiliangLi, 2017-06-19:
The system is observable if we fix the first pose.
Intuitively, when the second pose comes, it can be calculated by imu integration uniquely, then the features observed by 1rd & 2rd frame can be triangulated. 
Step by step.
If the camera moves in constant velocity when the imu data equals zero, the case you describe may happen.

closed 2017-07-17 by PeiliangLi