HKUST-Aerial-Robotics/VINS-Mobile #56: No viable overloaded '='

HungYaTsai, 2017-06-21:
Got an error in ViewController.m

if (!feature_tracker)
        feature_tracker = new FeatureTracker();

error message: no viable overloaded '='

Build on Xcode 9.0 Beta and ios 11

Do you think it might be the compatible problem?

PeiliangLi, 2017-06-21:
Actually, I have tested VINS-Mobile with Xcode 9.0 beta and iOS 11, but didn't meet the compile problem.
It should be ViewController.mm, have you changed the suffix？

HungYaTsai, 2017-06-21:
Thanks for the quick response, yeah it`s ViewController.mm, I didn`t change anything.

But since I already have boost,  I didn`t do the following step:

$ ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"
$ brew install boost

do you think it might be the problem?

HungYaTsai, 2017-06-21:
It works now, thanks again for your help

closed 2017-07-17 by PeiliangLi