HKUST-Aerial-Robotics/VINS-Mobile #59: Feature request: build a Google Tango version

greymfm, 2017-06-26:
Google Tango's own localization stack has a major drawback: it's a blackbox, e.g. you cannot optimize it or get localization confidence feedback (e.g. number of detected feature points etc.) and so it's useless for e.g. robotics projects...

Idea: make VINS a Google Tango version?

Here's how to get Google Tango sensor data (IMU+fisheye camera):
https://github.com/jinjaysnow/SenseData

Actually, the interesting part is in C++:
https://github.com/jinjaysnow/SenseData/blob/master/app/src/main/cpp/SenseDataApp.cpp

Do you think it's possible to create a VINS Google Tango version? :-)

I can help with Android programming if you like... Or I can make the Android app for you (I might need your help with the VINS part) - what do you think?

PeiliangLi, 2017-06-27:
Thanks for the concern.
Basically, the sensor driver parts are only included in ViewController.mm, I add some comments for better understanding it, please check the latest commit.
You can port other parts without many modifications as long as the third-party (opencv, ceres, Eigen) can be used on your platform.

greymfm, 2017-06-28:
jinjaysnow discovered that the initialization is really slow on his iPad Pro (https://github.com/jinjaysnow/SenseData/issues/1) - maybe the camera and IMU is not time-synchronized on these devices or what else could be the reason for it? I'm thinking how to build the cheapest device for VINS, and a Google Tango phone looks like one - will VINS work fast enough on it?