HKUST-Aerial-Robotics/VINS-Mobile #61: midPointIntegration issues in integration_base.h

kuonangzhe, 2017-06-29:
Hi, 

I went throught the function **midPointIntegration** inside integration_base.h. Though I know that the paper used Euler for easy representation, and the code is using mid-point integration for better accuracy. But the Matrix F and V really has huge differences from paper. The closed issue explanning 15*18 structure of six channel is still not clear for me. 

I also went through the TRO_technical_report and the paper "Tightly-Coupled Monocular Visual-Inertial Fusion for Autonomous Flight of Rotorcraft MAVs", unfortunately, they cannot help with the implemental difference and understanding. 

Could you mind providing some inference or the structure of the mid-point Matrix result? Or some other material which is available from you or online that might help with understanding this? You know, the large difference between code and paper really made it suffering to study this:D

Thank you so much!

kuonangzhe, 2017-06-29:
@PeiliangLi

qintony, 2017-07-02:
Please check out 
https://github.com/HKUST-Aerial-Robotics/VINS-Mono/issues/14

kuonangzhe, 2017-07-06:
@qintony Thanks for the help! Best regards!

closed 2017-07-06 by kuonangzhe