HKUST-Aerial-Robotics/VINS-Mobile #63: Jacobian question again

vonzy, 2017-06-29:
Hello, I have read paper _'Tightly-Coupled Monocular Visual-Inertial Fusion for Autonomous Flight of Rotorcraft MAVs'_ ,_'Technical Report VINS-Mono: A Robust and Versatile Monocular Visual-Inertial State Estimator'_ and _'IMU Preintegration on Manifold for Efﬁcient Visual-Inertial Maximum-a-Posteriori Estimation'_ carefully, and found there are some differences on jacobian calculation between the code and the papers. 
In paper _'Tightly-Coupled Monocular Visual-Inertial Fusion for Autonomous Flight of Rotorcraft MAVs'_ ,the jacobian for p,v,q part is 
[image]
but in the code, it's 
[image]
Also, in the paper 'IMU Preintegration on Manifold for Efﬁcient Visual-Inertial Maximum-a-Posteriori Estimation', the jacobians are
[image]
[image]
[image]
There still some differences for bias part. I wonder what cause the difference.
And I also noticed that in the code the jacobian for attitude is defined for two situations:
[image]
[image]
[image]
I don't know what is this for, could you please explain it?

vonzy, 2017-06-30:
@shaojie @qintony @PeiliangLi

qintony, 2017-07-02:
The difference is caused by the different definition of variables and residual function. I suggest you derivate it by yourself then you can know the difference behind them.

vonzy, 2017-07-02:
So why the jacobian of attitude is defined for two situation?

qintony, 2017-07-02:
The comment version is used when bias estimation is off.
We keep it in debug purpose, to valid the IMU correction is right.

closed 2017-07-17 by PeiliangLi