HKUST-Aerial-Robotics/VINS-Mobile #66: extrinsic transformation between imu and camera

1474123989, 2017-07-11:
How is the extrinsic transformation between imu and camera computed, while the paper is saying it is assumed to be known?

PeiliangLi, 2017-07-17:
Basically, 2 ways.
1. Collect camera and imu data, then calibrate them offline using kalibr https://github.com/ethz-asl/kalibr
2.  refer the datasheet and PCB，then measure them by hand.

1474123989, 2017-07-17:
Thanks! Another two related questions: It seesm in iOS the camera and imu sensor has the same frame (x/y/z has the same direction). Will it be the case on Android? And is it key for the calibration accuracy since Android world has so many different imu  sensors?

PeiliangLi, 2017-07-18:
1. The camera and imu on iPhone are not axis aligned, so they need extrinsic transform(R, T).
2. No.
3. Yes. So it's challenging to run a vio on general Android phones unless calibrating the intrinsic and extrinsic param online.

llpan91, 2017-08-02:
I have a question about extrinsic calibration between imu and camera .
if we use the kalibr package,  how to verify that the marked R and t are accurate?

TouqeerAhmad, 2017-10-27:
@llpan91 Hi, were you able to do extrinsic calibration between imu-camera for any ios device

zhixun25, 2020-07-20:
I would like to ask if there is a big gap between the external parameters of iPhone 7p calibrated by kalibr and the RIC and tic given by vins. Is the RIC in vins not a simple external parameter, but a rotation of the image is added

DanielMao2015, 2021-04-01:
@PeiliangLi  hi, I am new to slam and very appreciate for your excellent work. And I am confused by the extrinsic parameter calibration. As said in paper, the rotation and translation can be calibrated online, so why should we calibrated manully? Since it is very complex work for offline calibration.