HKUST-Aerial-Robotics/VINS-Mobile #69: VINS for android can't converge when huge rotation occurs

vonzy, 2017-07-22:
It's me again! :) 
I collect a dataset on android phone, and test with VINS. It performs very well at the beginning, but when huge rotation occurs, vision part fail to track feature points, and the result get diverged immediately.
[image]
I can't find what's wrong with that, please help me out, thanks!!!

FangGet, 2017-07-22:
Hi vonzy, can you share your no-ros vins project to me? I do not want to waste time do this part of work if you're kind to share. We can share android migrate progress with each other.

vonzy, 2017-07-23:
@FangGet Okay, don't you remember me?  I have seek for your help when I try to use your project  ORB_SLAM_Android.  The project is quite impressive, thanks for your help last time. But how can I send the project to you? through e-mail? or IM?

FangGet, 2017-07-23:
@vonzy , send me with the email, thanks.

trinamntn08, 2017-07-27:
@vonzy   :Hi, i'm working on testing VINS-SLAM. My goal is to collect data(sequence of images + IMU datas) to run off-line on VINS system. But i got a problem when i record data from Android device. In fact, i have used Samsung S6 to  record data in memory, but i just achieved to get 640x480 images with 15 Hz and 80 Hz in IMU datas. Because of that, it can't work well in VINS system. 
Do u have an ideal to resolve this issue ? 
I really appreciate your help.

PeiliangLi, 2017-07-27:
Hi, @vonzy VO is not magic, it's hard to get a good result when you give it pure rotation especially when you don't have the accurate extrinsic calibration.
Also, the image seems blurred seriously, you can try to reduce the exposure time.

vonzy, 2017-07-27:
@trinamntn08 , I haven't record data from S6. So I can only give you some advice based on your description. You should check:
1. Before you get sensor data, you must have registed the listener by call the function:
boolean registerListener(SensorListener listener, int sensors, int rate)
So have you set the rate to  SENSOR_DELAY_FASTEST? If so, have you put the listener into a seperate thread? Make sure the listener runs in a seperate thread. If you can 't still get higher rate data, please check whether your phone support higher rate with Sensor Kinetics.
2.Camera part is more unstable, it might varis from different OEM. First check whether the preview listener runs in a seperate thread, and check the support fps with getSupportedPreviewFpsRange(). If S6 support higher rate, you can try to set the fps with setPreviewFrameRate().

vonzy, 2017-07-27:
@PeiliangLi, okay, let me try, thanks!!

trinamntn08, 2017-07-28:
@vonzy thanks a lot for your help. I will do it.

JMZZZ, 2017-08-10:
@vonzy 
Hi, I also collect a dataset on android phone, but I covert data to a rosbag and test with VINS-Mono. The result is not good. I don't know how to solve this problem . I think you have  experience in this field. Could you please share with me ? In addition, I leave a message on your github, but you may  have no time to see it. Can you leave your email for convenience? I have some questions to ask you for advice.
My email is jmwhu@whu.edu.cn .Thank you very much!!!
[image]

vonzy, 2017-08-10:
@JMZZZ Sorry for no reply for your message. I rarely sign in github, you can contact with me at my email: von9604@gmail.com.

qintony, 2017-08-14:
@JMZZZ we have done some test on android...
The biggest problem is the unsynchronization. 
Apple devices do much better than Android in terms of synchronization.
I suggest you test the time offset between Camera and IMU firstly.

TouqeerAhmad, 2017-08-24:
@vonzy would you be able to share the android VINS -- I am trying to make it work for android as well.

vonzy, 2017-08-25:
@TouqeerAhmad I have only modified VINS-Mobile to a linux cmake version, there's no VINS for android yet. I am glad my work may help you, if you need linux version, please leave your e-mail address.

TouqeerAhmad, 2017-08-25:
@vonzy it would be great if you could share the VINS-Mobile for linux, it would be of great help. Thank in advance, here is my email.

sh.touqeerahmad@gmail.com

trinamntn08, 2017-08-28:
@vonzy  i'm working on VINS-mobile as well. Can you share with me your VINS-Mobile in Linux. I'll try to  make it work on Android. It would be a great help to me. Thanks a lot, my email : trinamntn08@gmail.com

trinamntn08, 2017-08-28:
@vonzy  i have question on the pipeline how you to test VINS on your data. In fact, if i understand, you recorded your data from Android device, you put all of them into a bag, and test VINS with this bag? because i'm really new in ROS, so i have a difficulty how to generate a bag with a right format as VINS does?

vonzy, 2017-08-29:
@trinamntn08 I have sent to you, please check your inbox.

wtmilk, 2018-01-08:
@PeiliangLi how can i test the time offset between Camera and IMU？Can you give some  suggestion?

jackyfriend, 2018-05-31:
@qintonguav would you please sent your android project of vins-mono to me? I have worked several days to compile the third-party libraries for android, but encounted a lot of problems. My email is rancherzhang@163.com. Thank you!

SueMM00, 2022-11-07:
> @qintonguav would you please sent your android project of vins-mono to me? I have worked several days to compile the third-party libraries for android, but encounted a lot of problems. My email is [rancherzhang@163.com](mailto:rancherzhang@163.com). Thank you!

Hi, I have been stuck in solving the compiling problem for a long time. Have you figured it out? Could you please send me your android project of vins-mono? Thank you very much in advance!

SueMM00, 2022-11-07:
> @qintonguav would you please sent your android project of vins-mono to me? I have worked several days to compile the third-party libraries for android, but encounted a lot of problems. My email is [rancherzhang@163.com](mailto:rancherzhang@163.com). Thank you!

@jackyfriend Hi, I have been stuck in solving the compiling problem for a long time. Have you figured it out? Could you please send me your android project of vins-mono? Thank you very much in advance!