HKUST-Aerial-Robotics/VINS-Mobile #74: How to export Camera+IMU data and VINS results?

danpeng2, 2017-08-04:
can not find about:
//#define RECORD_VINS true
//#define READ_VINS true
please help.

PeiliangLi, 2017-08-04:
You can add this define and find the corresponding code block by yourself.
It needs a little time to figure out how to use it.

erickingxu, 2017-09-14:
why to get IMU datas from device api? like  Motion data from ios device....

QingbiaoLi, 2017-10-24:
I am also interested to export the data the Camera+IMU from vins-mobile to certain data (.bag or others) so that we could do some work on that?

Plus do you have any surface reconstruction algorithm on VINS-mobile or VINS-mono?

zxp771, 2018-03-07:
@danpeng2 @QingbiaoLi Hi 
Did you guys work out with this issue?
I also try to export the camera and IMU date for vins-mobile.
If there is anything you know please tell me.
Thanks for your help.

srikar8, 2018-08-28:
@danpeng2 @QingbiaoLi @zxp771 
has any one solved this ?
I want to export camera and imu data from iphone.
thanks in advance.