HKUST-Aerial-Robotics/VINS-Mobile #75: Error in ceres/mutex.h

vivekarvind, 2017-08-07:
Hello,

While trying to build the app, I initially encountered this error:

`ceres/compressed_row_jacobian_writer.h file not found`

Then I searched for the file and added the file location "$(PROJECT_DIR)/VINS_ThirdPartyLib/ceres-solver/internal" to the **Header Search Paths** in the **Build Settings**. Now, this error is gone but I get another user-defined error in the **mutex.h** file in the same location. There error is 

`error Need to implement mutex.h for your architecture, or #define NO_THREADS`

I understand this error is based on the #define but no definition is being done in the previous lines. How to proceed? Is it recommended to change this part? No useful resources in the internet regarding this, neither did I find anything in the previous issues. Any help appreciated.

Thanks!

PeiliangLi, 2017-08-15:
Hi, are you using the same version ceres with we provide one?
Btw, have you chosen to build for your target?

vivekarvind, 2017-08-15:
Hi,

I clone using `git clone https://ceres-solver.googlesource.com/ceres-solver
`. You mean set the build configuration of the product to "Release"? If so, yes I'm setting it to Release.

PeiliangLi, 2017-08-16:
You don't need to download ceres solver by your self. Because it is already included in the project repo. Please use the ceres that we provided. Thanks.