HKUST-Aerial-Robotics/VINS-Mobile #78: Question about ProjectionFactor::sqrt_info

vonzy, 2017-08-15:
In VIN-Mobile, the ProjectionFactor::sqrt_info is set to FOCUS_LENGTH_X / 1.5 * Matrix2d::Identity(). It seems the sqrt_info should be two arbitrarily selected orthogonal bases, but why it should be set to the value FOCUS_LENGTH_X / 1.5 , could you please explain it?

vonzy, 2017-08-15:
convert to pixel error? but why divided by 1.5? @PeiliangLi @qintony

PeiliangLi, 2017-08-16:
sqrt_info is the sqrt of the information matrix, information matrix is the inverse of the covariance matrix.
We set 1.5 as the visual measurement noise in the pixel plane.

vonzy, 2017-08-16:
@PeiliangLi you mean the code assume pixel error is 1.5 pixel?

PeiliangLi, 2017-08-16:
yes

vonzy, 2017-08-16:
all right, thanks~ @PeiliangLi