HKUST-Aerial-Robotics/VINS-Mobile #79: box jumping and pose is unstable

ismymajia, 2017-08-17:
The box will jump a little when the phone is moved slightly. 

sometime the pose of box is not very stable.

i find that it is caused by r_drift in VINS.cpp.  Is there a better computer method?

PeiliangLi, 2017-08-18:
Thanks for the report.
It's because detecting loop and relocalizing frequently. To avoid this, you can increase the MIN_LOOP_NUM to reduce the loop recall rate. You can also increase the sleep time of the loop_thread when loop_succ is true.

ismymajia, 2017-08-21:
Thanks for your reply.
Fast relocation is required in mobile phone scene. So may need a better way for this problem!

HiZhongfei, 2017-08-26:
I also found the box would not be stable when the loop and relocalizing occurred, work even worse than the situation without loop. Is there possible some improvements in BA procedure when loop occured

bhadresh74, 2018-11-01:
Facing the same issue @PeiliangLi 
Let me know if you have any recent update on this. 

Thanks in advance.

ftahir999, 2019-05-22:
I am too getting the same issue, the boxes jump out of the scene and phone goes to detection phase quickly. @PeiliangLi @ismymajia did guys able to solve the issue?