HKUST-Aerial-Robotics/VINS-Mobile #82: Success to transplant VINS Mobile to Android platform, and test with MH_05_difficult

StevenCui, 2017-08-21:
Hi everyone,
I have transplanted the VINS-Mobile project to Android platform, and compared the result between the Android platform and ROS platform with the EuRoC MAV Dataset named MH_05_difficult.
In android project, the Initialization section could finish when the first descending of UAV, and the camera pose(translation and rotation) could be get. However，the pose is wrong!
The translation in vertical direction of ROS and android is showed below:
ROS: The result is similar with EuRoC dataset.
[image]
Android:
[image]
Do anyone know which issue went wrong?
If the loop detection and global pose graph optimization are ignored, and the local visula-inertial BA process is correct, but the initialization is not correct, will the path of the camera in android be simular with ROS's?
Thanks.
If anyone is working for it, welcome to talk about it together.

PeiliangLi, 2017-08-21:
Hi, there must be something wrong or missed during your porting, because too many details need to be considered. It's hard to debug with only 2 figures...

vonzy, 2017-08-21:
@PeiliangLi is right, there are too many details need to be considered.....

PeiliangLi, 2017-08-21:
Maybe @vonzy can help you, I think he has experience on this.

vonzy, 2017-08-21:
@PeiliangLi ..... I am still struggling with time synchronization, and can't work it out. Do you have any idea?

PeiliangLi, 2017-08-21:
1.  Rotate your phone and keep the camera face to the chessboard meanwhile.
2.  Collect the image data and imu data with timestamp.
3.  Calculate the pitch and roll rotation from image(solve pnp) and acc data separately.
4. Align the rotation curve in matlab, then you can find the time offset and if the offset is constant.

vonzy, 2017-08-21:
@PeiliangLi calibrate the delay offline? sounds great! But why must face to chessboard?

StevenCui, 2017-08-21:
@PeiliangLi @vonzy Thanks for your reply. I know there are some problem during my porting to android. I want to locate which part the problem may be in.
Because there are so many details to discuss, i will list my questions to ask your help:
1. For MH-05, the external parameters are [0 -1 0; 1 0 0; 0 0 1] in ROS config，what is the corresponding roll/pitch/yaw in ios?
2. Do u know  anyone has tested the MH-05 in the VINS IOS? Does it work well?

vonzy, 2017-08-22:
@StevenCui 
1. the extrinsic parameters is  [0 -1 0; 1 0 0; 0 0 1] for all euroc dataset indeed, no matter  which platform you run the program on. If you want to test the vins on android with android phone data, you should use android extrinsic parameters, which is [0 -1 0; -1 0 0; 0 0 -1].
2. No.

StevenCui, 2017-08-22:
@PeiliangLi I changed the SOLVER_TIME that is used in VINS::solve_ceres, from 0.06s to 0.6s. I find the camera pose is different, like this:

StevenCui, 2017-08-22:
Supplement: For MH-05, the termination of ceres solve in non-linear optimization is CONVERGENCE on Android, but it is NO_CONVERGENCE on ROS.

StevenCui, 2017-08-23:
@PeiliangLi Hi, have you tested the  VINS-Mobile project on iphone with MH-05? Does it work well?

StevenCui, 2017-08-28:
I create a QQ group named "港科大VINS讨论群"，which num is 634995296.Welcome to join us!

TouqeerAhmad, 2017-08-30:
Have anybody tried with S8 samsung -- can anyone please let me know the extrinsics between camera and imu

bileki, 2017-09-21:
@StevenCui, your version for android is available?

TouqeerAhmad, 2017-10-27:
@PeiliangLi I am trying to collect some data from a mobile device and then run the VINS offline using this data. This is what I have been trying: Get the timestamped date from iphone -- both imu and camera. Then perform imu-camera calibration using Kalibr and then use VINS with this calibrated file and offline data from iphone. I just read these comments from you:

"Rotate your phone and keep the camera face to the chessboard meanwhile.
Collect the image data and imu data with timestamp.
Calculate the pitch and roll rotation from image(solve pnp) and acc data separately.
Align the rotation curve in matlab, then you can find the time offset and if the offset is constant."

Can you please elaborate a bit more on each of the points. I think I am messing up the synchronization for imu-camera and that is what messing my calibration. Any help in this regards would be great!