HKUST-Aerial-Robotics/VINS-Mobile #83: Opencv Problem when porting to Android

xiaodian2234, 2017-08-22:
I have porting VINS to Android but I find that the performance of opencv on Android is much lower than IOS. 
Now I have difficulty in using calcOpticalFlowPyrLK on Android, which cost around 12ms on Android but less than 5ms on IOS.   I think this is probably because multithreading on Android is not high-performancly supported as on IOS. 
Could you suggest some method or other library that can realize Optical Flow on Android? Thanks a lot.

sandmen, 2017-08-22:
@xiaodian2234,
on Android, you have very different results from device to device. :-(
Apple as the advantage, to support only "one" HW.
Do you have anywhere your port of VINS, so, I can check it on my device... ?

StevenCui, 2017-08-22:
Hi，i am porting to android too, and i am testing the accuracy of android project with MH-05 dataset. Do u have tested it with dataset?

xiaodian2234, 2017-08-23:
@sandmen I have tested on different Android phones. It seems that higher hardware configuration can get better result.

xiaodian2234, 2017-08-23:
@StevenCui No. I only test it online or by recording data in our environment.

TouqeerAhmad, 2017-08-24:
@xiaodian2234 Hi, do you have any plans to make the android version public ??

shopkauf, 2017-08-25:
@xiaodian2234 : would you mind to share your android version with me? Thank you!

FangGet, 2017-08-29:
opencv has an 3rdpary named carotene where opticalflow calculation is conducted on NEON, just have a try with it @xiaodian2234

xiaodian2234, 2017-08-30:
@FangGet I have tested it. The average time of LK has reduced about 5ms. But the main problem in our Android phone is that sometimes the course can jump to about 20ms . The procedure seems not stable on current platform. We think this may be caused by cpu schedule.

bileki, 2017-09-21:
@xiaodian2234 your android version is online? I would love to test on my motoG1 to check the performance as I am getting the data from motoG1 and doing the calculations on a server.