HKUST-Aerial-Robotics/VINS-Mobile #92: VINS iPhone 6

hjwdzh, 2017-10-13:
If I want to use iPhone 6, do I only need to provide the intrinsic and extrinsic parameters (fx, fy, ppx, ppy, tic_xyz)? What is tic_xyz BTW?

vonzy, 2017-10-19:
@hjwdzh check global_param.hpp and global_param.cpp.
You can provide intrinsic parameters by setting values to  FOCUS_LENGTH_Y, FOCUS_LENGTH_X, PY, PX;
and provide extrinsic parameters by by setting values to TIC_X,TIC_Y,TIC_Z.
tic_xyz is the vector from imu center to camera center, in latex it can be written as t^i_c

zxp771, 2018-02-17:
Hi,hjwdzh @hjwdzh
have you build the app in your iPhone 6?
I have some problem about building the app.

Solin32, 2022-03-16:
> @hjwdzh check global_param.hpp and global_param.cpp. You can provide intrinsic parameters by setting values to FOCUS_LENGTH_Y, FOCUS_LENGTH_X, PY, PX; and provide extrinsic parameters by by setting [...]

Would you like to verify that TIC is actually the vector from camera to IMU, not IMU to camera?