HKUST-Aerial-Robotics/VINS-Mobile #93: Max value of the "freq:" parameter in the config file

SubMishMar, 2017-10-17:
What can be the maximum value of "freq:" parameter in the config file? I think it is limited by the frequency of the incoming image topic and if I give any number greater than that, then it will just publish the feature images at the same frequency as the incoming images. Please correct me  if I am wrong.

PeiliangLi, 2017-10-26:
It's determined by Apple API.
According to my test, the max frequency of raw image is 30 HZ. And the max frequency which vins can process in real-time is also 30 HZ.

SubMishMar, 2017-10-26:
Thanks for the response..,. sorry for posting here, I am actually asking about VINS-Mono not VINS-Mobile. Can I run VINS mono with the camera feed coming in at 60 Hz and the feature tracking working at same frequency?
Just FYI, my imu freq is roughly 230 Hz and my image topic freq is 60 Hz.