HKUST-Aerial-Robotics/VINS-Mobile #95: Build problems in Xcode

Austin1511, 2017-10-23:
Hi guys,

I followed the instruction you give me to build and run the project in Xcode, but I still got these errors. Do you know what is happening? BTW, I'm using the Xcode 9.0.
[image]
[image]
Thank you!

QingbiaoLi, 2017-10-24:
Make sure that you download opencv.framework from the given link and unzip it to ''VINS-moblie/VINS_ThirdPartyLib/''
So that in 'VINS_ThirdPartyLib' folder, there is 'opencv2.version' and 'opencv2.framework'. Then it should be fine.

HustLion, 2017-10-31:
The library provided by the given link is still problematic. I compared it with standard opencv2 release, and 
[image]

and those *.hpp files in `Headers-1` should be copied to `Headers` folder so that `<opencv2/core.hpp>` can be found.

After solving this, now it's complaining about libceres, trying to solve this now.

YuehChuan, 2018-03-30:
Thank you @HustLion ! the tricks work!