HKUST-Aerial-Robotics/VINS-Mobile #96: Data Collection App

TouqeerAhmad, 2017-10-26:
Hi Everyone, I was wondering if anyone has successfully tested the VINS-Mobile/VINS-Mono on offline mobile data. I am looking what steps to follow to get the synchronized imu/images from a mobile device; if any of you have aware of such an app, please let me know. 

I have been trying with Samsung S8 but so far have not been successful to run VINS-Mobile on this data. I used Kalibr for imu-camera calibration and used data sheet to find the noise densities etc. for the imu.yaml file.

Any help in this regards will be great. Thanks!

PeiliangLi, 2017-11-10:
I have a repo: https://github.com/PeiliangLi/ios2ros which convert image and imu txt file into a ros bag. Then you can run VINS-Mono using the bag easily.

zxp771, 2018-02-17:
Hi,Peiliang @PeiliangLi 

cause I failed to run vin-mobile on my phone.
I'm wondering is there any way that I can test my own dataset(like record a short video by my phone)
on VINS-Mono?

Thanks for your help!

EmreOzkose, 2020-03-26:
Hi ! @PeiliangLi  In VINS-MONO readme, **The image should exceed 20Hz and IMU should exceed 100Hz** is written. However, I receive images with 1Hz in my Android App. Is it a problem?