HKUST-Aerial-Robotics/VINS-Mobile #98: How about capture high-solution image

loagingray, 2017-11-09:
Hi there,

I would like to capture a 1920x1080 picture for other use before vins,
and then use cv::resize to make it fit your program 640x480,
will it came up with problem?

I cannot find whether the position vins give is right,
But it seems abnormal things happen more frequently that the camera in vins move far away but in fact I sit still.
Could someone help?

vonzy, 2017-11-10:
won't.
You should check whether imu noise parameters, imu-camera extrinsic parameter are correct. And make sure that imu is synchronized with camera, and the frequency of sensor data is steady.

loagingray, 2017-11-10:
Thanks vonzy,
in fact I realize it may came up some problem using resize directly,
I try to resize 1920x1080 to 640x360 and then paste it to 640x480 image with black on both side.
The image I got is almost the same with capture 640x480 image except black strip on the both side.
And after then, vins work properly as before. 

Btw, I am curious about how you get Device parameter in global_param.hpp.

PeiliangLi, 2017-11-10:
Hi, @loagingray The first thing you need to do is collecting the resized images and calibrating the camera intrinsic parameters rather than using the default one.

vonzy, 2017-11-10:
PeiliangLi is right, correct camera intrinsic parameters are essential for vins also, you can use matlab toolbox to get them from resized image captured from chessboard calibration.

loagingray, 2017-11-10:
Hi, @PeiliangLi  & @vonzy 
I got it!
Thx for help!