HKUST-Aerial-Robotics/VINS-Mobile #100: VINS-Mobile always appear "FAIL_COST".

Chenwei-Jheng, 2017-11-15:
I try VINS_Mobile to iPhone 6 Plus, but STA Label appears FAIL_COST.
The iPhone 6 PLUS params are 
            FOCUS_LENGTH_X = 547.565;
            FOCUS_LENGTH_Y = 547.998;
            PX = 239.033;
            PY = 309.452;
            SOLVER_TIME = 0.08;
            FREQ = 3;
            //extrinsic param
            TIC_X = 0.0;
            TIC_Y = 0.065;
            TIC_Z = 0.0;

vonzy, 2017-12-14:
Maybe not, the optimization doesn't require camera intrinsic parameters of high accuracy （although the result will be worse, but no failure. you should check the quality of the image or the relative rotation between imu and cam.

Chenwei-Jheng, 2018-01-03:
In Xcode, I set Edit Scheme -> Run -> Info -> Build Configuration for Release.
It is successfully. Thanks.

kismeter, 2018-11-14:
@Chenwei-Jheng Does the issue solved? I also met this kind of issue, always appear "FAIL_PARA" "FAIL_COST" "FAIL_SFM" and "FAIL_ALIGN". when porting to Android

Chenwei-Jheng, 2018-11-14:
Maybe the Ceres solver is not accurate in Android.
In Xcode, I set "Optimization Level" for "Faster [-O2]", the VINS is successful.
You can try to find this Setting in Android Studio or eclipse.

The "Optimization Level" is finded for "Xcode" -> "Build Setting"->"Apple Clang - Code Generation"->"Optimization Level".

kismeter, 2018-11-14:
> Maybe the Ceres solver is not accurate in Android.
> In Xcode, I set "Optimization Level" for "Faster [-O2]", the VINS is successful.
> You can try to find this Setting in Android Studio or eclipse. [...]

I explicitly set build type to Release, and change Optimization level for O2, the VINS successful. Thanks

`-DCMAKE_BUILD_TYPE=Release`