HKUST-Aerial-Robotics/VINS-Mobile #103: Some problem about porting to android.

ray2004a, 2017-12-28:
Thanks for the wonderful sharing. I am trying porting VINS-Mobile to android system. And first collected some data from Samsung NOTE5. for 120sec data( 50sec on starting point and only slowly rotatoing camera and 70sec moving slowly). But here got some problem below:

1.In initial_aligment.cpp->SolveScale, fabs(g.norm() - G_NORM) can't be under G_THRESHOLD   --> [ solve g failed!]
2. PS.z value keep increasing even I didn't move but just rotate the camera. (like growing to 20000~30000up)

3."Ceres Solver Report" on initial_sfm is always be Termination: NO_CONVERGENCE. but sometimes it still can show "vision only BA converge".

Here are the IMU parameter and camera parameter I used:
[IMU]
accelerometer_noise_density: 0.1 
accelerometer_random_walk: 0.01 
gyroscope_noise_density: 0.05 
gyroscope_random_walk: 4.0e-5

[Camera]
FOCUS_LENGTH_X = 1.1148352059937931e+003;
FOCUS_LENGTH_Y = 1.1148352059937931e+003;
PX = 319.0;
PY = 239.0;            
SOLVER_TIME = 0.06;
FREQ = 3;            
TIC_X = 0.0;
TIC_Y = 0.0;
TIC_Z = 0.0;

The IMU frequency is 200HZ, Camera frequency is about 12.5HZ( limitation of mobile storage speed).
It will be very thanks to any suggestion.

ray2004a, 2017-12-28:
System log can be downloaded [here](https://drive.google.com/open?id=17Wtw8hYV4UhiZw5haldAMBZrr95Im8k-):

thanks for any advise.

PeiliangLi, 2018-01-29:
The acc measurements should be **m/s^2**, but what you used is **g**

dvlee1024, 2018-08-29:
@ray2004a can you share your android version vins?

Charleshzhang, 2018-08-30:
@ray2004a I just started working on porting VINS to android too, do share plz

Charleshzhang, 2018-08-30:
@PeiliangLi Did you guys downsized the ceres lib? I tried to compile ceres, got a static lib of 618M, while your static lib is only 16M. Any idea?

zhixun25, 2020-07-21:
> The acc measurements should be **m/s^2**, but what you used is **g**
Hello, I would like to ask you that the measurement of ACC should be m / S ^ 2. Could you elaborate on what m and s represent. Because I want to run the euroc dataset on IOS, but always align fails. The internal parameters, external parameters and IMU noise parameters are all given by the euroc. Anyway, thank you very much

ouceduxzk, 2021-11-07:
https://github.com/ouceduxzk/VINS_android , this is a github repo that have vins mobile running on android

dandingol03, 2022-11-28:
> https://github.com/ouceduxzk/VINS_android , this is a github repo that have vins mobile running on android

no,it cannot run on android studio