HKUST-Aerial-Robotics/VINS-Mobile #104: VINS-mobile by iPhone 8

KangTeng, 2018-01-09:
I obtain intrinsic parameters such as FOCUS_LENGTH_Y, FOCUS_LENGTH_X, PY, PX and get extrinsic parameters likes TIC_X,TIC_Y,TIC_Z by monocular camera calibration. After the compilation is successful, run the code on the IPhone8, the interface stays at 0% of the initialization (Feature points can be refreshed).

dreness, 2018-01-29:
@kangteng - would you mind sharing your iphone8 camera intrinsics and a brief description of how you obtained them? I am currently trying to get this project working with my 8+, and although I have found a couple ways to derive the intrinsics, the results aren’t as consistent as I’d like...

KangTeng, 2018-01-29:
I can run VINS-mobile successful on the IPhone8 and IPhone8+, you can try the intrinsics of IPhone8+ like this:FOCUS_LENGTH_X=530.233;FOCUS_LENGTH_Y=531.082;PX=250.286;PY=316.520.(I don't use the raw data, however, it is obtained by multiplying  a factor by Comparing the intrinsics of iphone7 in the source code  and the measured data by monocular camera calibration)

Chun-iii, 2019-02-26:
I have built the app successfully in my iPhone8. But it shows unsupported device and the initializing is always 0%. How did you solve this problem?