HKUST-Aerial-Robotics/VINS-Mobile #105: VINS-Mobile in iPhone 6s

zxp771, 2018-02-13:
can't finish compile.the issues is strange.I never see the same problem from others.

"Check dependencies

Code Signing Error: No account for team "59LPDRL4NH". Add a new account in the Accounts preference pane or verify that your accounts have valid credentials.
Code Signing Error: No profiles for 'mobile.VINS-ios' were found:  Xcode couldn't find any iOS App Development provisioning profiles matching 'mobile.VINS-ios'.
Code Signing Error: Code signing is required for product type 'Application' in SDK 'iOS 11.2'

"

zxp771, 2018-02-17:
@PeiliangLi

closed 2018-02-19 by zxp771

YuehChuan, 2018-03-30:
@zxp771  this about create a iOS developer user!
signing and Bundle identifier need to create a user.
check  [this](https://medium.com/彼得潘的-swift-ios-app-開發問題解答集/從-xcode-8-將app安裝到-iphone-的小小2個步驟-39f7b81b69a6)

Second you may encounter opencv  dependency error 
follow #95 

Good luck!