HKUST-Aerial-Robotics/VINS-Mobile #119: iPhone SE support

naserkhosravi, 2018-07-11:
I can't see iPhone SE in supported device!
Is there any technical reason ?
Thanks.

se7oluti0n, 2018-08-25:
Hi @naserkhosravi , I would like to share the code modification to work with iphoneSE

in function `DeviceType deviceName()` in file `ViewController.mm`, add following code
```C++
    else if(([device compare:@"iPhone8,4"] == NSOrderedSame))
    {
        printf("Device iPhoneSE\n");
        device_type = iPhoneSE;
    }
```
in file `global_param.hpp`, add `iphoneSE` to the `enum DeviceType`:
```C++
enum DeviceType
{
    iPhone7P,
    iPhone7,
    iPhone6sP,
    iPhone6s,
    iPhoneSE,
    iPadPro97,
    iPadPro129,
    unDefine
};
```

in file `global_param.cpp`, add iphoneSE calibration data to function `setGlobalParam`:
```C++
        case iPhoneSE:
            printf("Device iPhone6s param\n");
            FOCUS_LENGTH_Y = 529.90;
            PY = 315.30;
            FOCUS_LENGTH_X = 527.63965351;
            PX = 239.468;
            SOLVER_TIME = 0.06;
            FREQ = 3;
            
            //extrinsic param
            TIC_X = 0.00017249;
            TIC_Y = 0.0003245;
            TIC_Z = 0.00002171;
            return true;
            break;
```

Hope this help

naserkhosravi, 2018-08-25:
@se7oluti0n Very thanks, this works very good.
I did not see above parameters until now, how can obtain above parameters?
A clue is enough for me.
Thanks again.

closed 2018-08-25 by naserkhosravi

se7oluti0n, 2018-08-26:
@naserkhosravi 
where do you get these value 
```
FREQ = 3;
TIC_X = 0.0;
TIC_Y = 0.092;
TIC_Z = 0.1;
```

basically, you can get the calibration value by doing the camera-imu calibration, using kalibr calibration tool, https://github.com/ethz-asl/kalibr, following example steps:
1. record camera images and imu data from iphone while capturing the calibration board from kalibr. 
2. Convert that data to rosbag
3. run the camera calibration and camera-imu calibration with kalibr

naserkhosravi, 2018-08-26:
@se7oluti0n Very thanks for your help.

lishengling1, 2022-03-28:
I can't see iPhone 13 pro max in supported device!
Is there any technical reason ?

else if(([device compare:@"iPhone8,4"] == NSOrderedSame))
    {
        printf("Device iPhoneSE\n");
        device_type = iPhoneSE;
    }
here,why IPhone8,4 is iPhoneSE? my phone is iphone 13 pro max,what is my device compare?
Thanks.