HKUST-Aerial-Robotics/VINS-Mobile #124: 一直Initializing:0%的问题，and修改了几处可能的逻辑问题和c11用法问题。

xx8086, 2018-11-28:
您好，我昨天拉去的最新代码。ip6s(ios11) + mac10.12.6(xode9.0)
能画出红、蓝点，但是一直Initializing:0%，无法画出网格和cube。

我这边getMeasurements函数一直wait for imu,或者throw img返回size ==0，导致process之后的代码不能执行。

不好意思不太懂oc简单看了下。
我改了些运行报错和代码逻辑可能有问题的地方：
1，
- (void)imuStartUpdate{
    CMMotionManager* motionManager = [[CMMotionManager alloc] init];
改为
- (void)imuStartUpdate{
    //CMMotionManager* motionManager = [[CMMotionManager alloc] init];

and
        [self imuStartUpdate];
        isCapturing = YES;
        motionManager = [[CMMotionManager alloc] init];
        [mainLoop start];
改为
motionManager = [[CMMotionManager alloc] init];
        [self imuStartUpdate];
        isCapturing = YES;
        [mainLoop start];

2，
FeatureTracker::FeatureTracker()
:mask{ROW, COL, CV_8UC1},
改为
FeatureTracker::FeatureTracker()
:mask(ROW, COL, CV_8UC1)/*mask{ROW, COL, CV_8UC1}*/,

xx8086, 2018-11-30:
[image]
img_msg_buf.push(img_msg);
[image]

PeiliangLi, 2018-11-30:
The reason might be you didn't use the provided opencv framework.

[a78c453a29e01b0d535dcdf6f2e4d56e]<https://user-images.githubusercontent.com/3690609/49271212-a46b2080-f4a7-11e8-8cbb-134303e8072d.png>

[6f781220430350ca3d4a896c4b60cf71]<https://user-images.githubusercontent.com/3690609/49271220-a8973e00-f4a7-11e8-9e4b-42474982d9f0.png>

xx8086, 2018-11-30:
[image]
 hi, im use the provided opencv framework. 
but init75%, jacobian_pose_i.maxCoeff() < 1e8

xx8086, 2018-11-30:
hi, im use the provided opencv framework.
but init75%, jacobian_pose_i.maxCoeff() < 1e8

xx8086, 2018-11-30:
3q!!!
 I succeeded after doing these operations ：
cd /usr/local/include
rm -rf opencv
rm -rf opencv2
cd /usr/local/lib
rm -f libopencv*

closed 2018-11-30 by xx8086