HKUST-Aerial-Robotics/VINS-Mobile #125: Signing problem

pmd2245, 2018-11-29:
Hi,

I am novice with Xcode. This is probably a stupid question but I turn around a "signing" problem that I am not able to solve.
[image]

I would really appreciate if someone could help me!

Is there anyone who managed to test the app on iPhone X? 

Thanks in advance,
Pm

btbufresh, 2018-12-18:
Hi, any progress on the issue above? I met the same problem as you, and still haven't solved it. Help needed...

pmd2245, 2018-12-18:
Hi,

Unfortunately I did not find any solution. 
If you figure out something, please let me know.

Thanks,
Pierre-Marie

pmd2245, 2018-12-19:
You must change the Bundle Identifier. I just deleted the dash and now it works well.
[image]

closed 2018-12-19 by pmd2245