HKUST-Aerial-Robotics/VINS-Mobile #133: when and where is the function processImage of ViewController.mm called?

IvyGongoogle, 2019-09-17:
Hello, I am a novice and do not know when and where the function `processImage` of ViewController.mm is called. 
Can you give some advises ? Thanks.

IvyGongoogle, 2019-09-18:
need help..

NumenZ, 2020-05-30:
@IvyGongoogle 
It's a callback from OpenCV camera module.
Basically, as soon, as you declare a camera delegate
```obj-c 
self.videoCamera.delegate = self;
```
and start capturing
```swift
[videoCamera start];
```
instead of default __AV__ output style, you are going to receive images in OpenCV format(i.e. `cv::Mat` with 4 Chanels(`BGRA`)).

However one detail here, is that in current work it appears a _patched_ version of OpenCV seems to be used. 
```
float lowPart = image.at<float>(0,0);
float highPart = image.at<float>(0,1);
```
Original function, doesn't store timestamps on the borders of images(and all code further seems to take that in account).