HKUST-Aerial-Robotics/VINS-Mobile #140: Link VINS-ios-framework (arm64): Undefined symbols for architecture arm64 (NOT THE OLD PROBLEM ABOUT x86_64)

Solin32, 2021-06-07:
I have read all the issues, and find the only similar problem "Build Framework failed #45". I tried the solution in #45, but it didn't work. I don't have the problem about "Undefined symbols for architecture x86_64 "

Here are the errors:
[image]
[image]
[image]

environment:
MacOS 11.2.3
Xcode 12.5
Boost 1.76.0
opencv2.framework is the provided version
target device: iPad Pro 3 (11)

This problem has taken me a very long time, and I'm very depressed... Thank you if you could take a look! Thank you very much!

zymale, 2022-05-15:
@ShurlinZhao   your problem been solved? I have the same problem on iPhone 13.

KunruiHuang, 2022-12-01:
Hello, I have encountered the same problem and solved it. You can add the source file of the error reporting function in the compilation settings.